FreeCAD/FreeCAD #19286: Assembly: Seg fault when swithing from rigid to flexible sub assemblies.

oficinerobotica, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using sub assemblies inside a main root assembly , changing the "rigid" property to false will give an error "Assembly#Joint001 is unconnected to a grounded part so it is ignored."  This is wrong because the subassembly is connected to a grounded part via a fixed joint.

Re changing the rigid property to true will seg fault the app with the attached message .

Steps to reproduce:
In the attached file change the property "rigid" for the Assembly001 from false to true and then again to false.

https://github.com/user-attachments/assets/7797e770-9fae-49bb-9903-224e04f57cd6

```
FreeCAD 1.1.0, Libs: 1.1.0devR39980 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

NOTICE: Addon "/home/flo/.local/share/FreeCAD/Mod/sheetmetal.backup1729482375.2960367" disabled by presence of ADDON_DISABLED stopfile
(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
QFont::fromString: Invalid description 'SF Pro Text,13,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Hack,12,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'SF Pro Text,13,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'SF Pro Text,13,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
NOTICE: Addon "/home/flo/.local/share/FreeCAD/Mod/sheetmetal.backup1729482375.2960367" disabled by presence of ADDON_DISABLED stopfile
Assembly#Joint001 is unconnected to a grounded part so it is ignored.
Assembly#Joint002 is unconnected to a grounded part so it is ignored.
Assembly#Joint003 is unconnected to a grounded part so it is ignored.
Assembly#Joint001 is unconnected to a grounded part so it is ignored.
Assembly#Joint002 is unconnected to a grounded part so it is ignored.
Assembly#Joint003 is unconnected to a grounded part so it is ignored.
3.1e-08 <App> Document.cpp(3211): Assembly#Joint001 still touched after recompute
2.9033e-05 <App> Document.cpp(3211): Assembly#Joint002 still touched after recompute
3.5397e-05 <App> Document.cpp(3211): Assembly#Joint003 still touched after recompute
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x74c158445320]
#1  0x74c0f50579d3 in Assembly::AssemblyLink::onChanged(App::Property const*) from /tmp/.mount_FreeCABDHlPI/usr/lib/AssemblyApp.so+0x3d3
#2  0x74c15a79e545 in App::Property::hasSetValue() from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADApp.so+0x145
#3  0x74c15a7ae488 in App::PropertyContainerPy::setCustomAttributes(char const*, _object*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADApp.so+0x298
#4  0x74c15a7aeba3 in App::PropertyContainerPy::_setattr(char const*, _object*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADApp.so+0x13
#5  0x74c15a19565b in Base::PyObjectBase::__setattro(_object*, _object*, _object*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADBase.so+0x10b
#6  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(PyObject_SetAttr+0xfb) [0x74c15addce1b]
#7  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x3f6c) [0x74c15adf4dac]
#8  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x74c15aeabd1d]
#9  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x74c15aeab794]
#10  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x74c15aecabf3]
#11  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x74c15aec669a]
#12  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x74c15aeba9c9]
#13  0x74c15a14779c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADBase.so+0x6c
#14  0x74c15b7c4df4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x94
#15  0x74c15bb1c0a7 in Gui::PropertyEditor::PropertyItem::setPropertyValue(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x357
#16  0x74c15bb1e9de in Gui::PropertyEditor::PropertyBoolItem::setValue(QVariant const&) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x8e
#17  0x74c15bafe4c7 in Gui::PropertyEditor::PropertyItem::setData(QVariant const&) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x37
#18  0x74c15bb25c1c in Gui::PropertyEditor::PropertyModel::setData(QModelIndex const&, QVariant const&, int) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x3cc
#19  0x74c15bb2369d in Gui::PropertyEditor::PropertyItemDelegate::setModelData(QWidget*, QAbstractItemModel*, QModelIndex const&) const from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0xdd
#20  0x74c159bc3087 in QAbstractItemView::commitData(QWidget*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0xa7
#21  0x74c15baec8f6 in Gui::PropertyEditor::PropertyEditor::commitData(QWidget*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x26
#22  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x74c158aeb964]
#23  0x74c159bdf32f in QAbstractItemDelegate::commitData(QWidget*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0x2f
#24  0x74c15bb23845 in Gui::PropertyEditor::PropertyItemDelegate::valueChanged() from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libFreeCADGui.so+0x35
#25  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x74c158aeb6f2]
#26  0x74c159a6ba01 in QComboBox::activated(int) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0x31
#27  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5(+0x26cf00) [0x74c159a6cf00]
#28  /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x74c158aeb964]
#29  0x74c159a6bc12 in QComboBoxPrivateContainer::itemSelected(QModelIndex const&) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0x22
#30  0x74c159a6c4ae in QComboBoxPrivateContainer::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0x44e
#31  0x74c158ab68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Core.so.5+0x7b
#32  0x74c15997039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCABDHlPI/usr/bin/../lib/libQt5Widgets.so.5+0x6e
[... 36 lines cut ...]
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.36
  * PieMenu 1.11.2
  * OpenTheme 2024.9.1
  * freecad.gears 1.3.0
  * sheetmetal.backup1729482375.2960367 0.5.1 (Disabled)
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * Assembly3 0.12.2
  * Gridfinity 0.8.0
  * ArchTextures
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-01-27:
Related issue with flexible/rigid assembly toggle: https://github.com/FreeCAD/FreeCAD/issues/17903 Not a crash (different symptoms), but there is some discussion about the `[...] is unconnected to a grounded part so it is ignored` message as well.

oficinerobotica, 2025-01-27:
@maxwxyz It happens on the latest weekly build .... did I reported something wrong?

maxwxyz, 2025-01-27:
@oficinerobotica no sorry, misread the about info. Thanks!

oficinerobotica, 2025-07-24:
@PaddleStroke  Unfortunately on build 42653 I still get the crash and the "unconnected to a ground joint" message switching from the rigid property on and off. Should I wait for the next build?

PaddleStroke, 2025-09-24:
Found the cause of the crash.

closed 2025-09-24 by chennes