FreeCAD/FreeCAD #19289: Process does not match instruction manual

JCS22356, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Following manual instructions for the Lego block - attempting to constrain lines per instruction - dialog at bottom states impossible constraint.

### Full version info

```shell
FreeCAD 1.0.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

Attempted to add a file - file shows up in my desktop folder, but does not show up using 'add file' below

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-28:
Please explain at which step this occurs for you.

JCS22356, 2025-01-28:
Thank you for responding - Pg 81 of the manual - issue starts with last bullet - it is not obvious to me that constraints have been added - are the instructions missing some steps?

Page 82 first bullet   ". . . Add three more constraints . .  ." This triggers the 'impossible message'.
Also the screen does not look like this manual picture - it is white.

I have tried reloading the program - no change.

I have extensively used earlier 3D Macintosh programs successfully, so 3D is not new to me.  Jim

Roy-043, 2025-01-29:
I can't reproduce what you describe. The Wiki page has been updated recently, which may explain this. But there are some omissions and mistakes in the instructions. It is important that you create the rectangle without clicking on either of the axes. You should also not enter On-View-Parameters. The instructions for the symmetry constraint are faulty, the order of the selected points should be different.

https://wiki.freecad.org/Manual:Modeling_for_product_design
https://wiki.freecad.org/Sketcher_Workbench#On-View-Parameters

Roy-043, 2025-01-29:
I have updated the discussed section of the Wiki page.

JCS22356, 2025-01-30:
Thank you - I’ll review and try again - JimSent from my iPad

closed 2025-09-27 by Roy-043