FreeCAD/FreeCAD #19290: Cannot remove body from Part contained in array

benj5378, 2025-01-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Cannot remove body from Part contained in array.

https://github.com/user-attachments/assets/c0271b34-5dd7-4a9b-8d9e-7d9f92e3b561

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39986 (Git) Snap 1283
Build type: Release
Branch: main
Hash: 6d64dcfd88716840b4a7bd3428b5bc28a668a637
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-28:
You mean only by drag & drop?

Roy-043, 2025-01-28:
Confirmed. It only occurs for Link arrays. So it is probably a Core issue.