FreeCAD/FreeCAD #19293: Errors when editing additional parts

DrMoriarty, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello! FreeCad seems very interesting, but when I tried to make my first model I faced some problems. I made on object using additive and substractive bodies and try to edit ones to archive the desired state.
After that my model became invalid and my changes not appeared on the 3d view. Also I see some errors on the bottom of screen:
```
08:15:23  The mesh data structure has some defects
08:21:39  Adding face failed on Face15. Omitted
08:21:54  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.NeutralPlane missing element reference phone_cap#Cylinder003 ;#2b7:3;:G#2f7;RFI;:H94e:b,F.Face5
08:21:54  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face5
08:21:54  Draft: Invalid shape name ?Face5
08:22:18  Adding face failed on Face15. Omitted
08:25:22  <PropertyLinks> PropertyLinks.cpp(429): phone_cap#Draft.NeutralPlane auto change element reference phone_cap#Cylinder003 ;#2b7:3;:G#2f7;RFI;:H94e:b,F.Face5 -> ;#2b7:3;:G#3c3;RFI;:H94e:b,F.Face5
08:25:23  Adding face failed on Face15. Omitted
08:25:34  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.Base missing element reference phone_cap#Cylinder003 ;#2a0:1;:H94e,F.Face12
08:25:34  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.NeutralPlane missing element reference phone_cap#Cylinder003 ;#2b7:3;:G#3c3;RFI;:H94e:b,F.Face5
08:25:34  <Exception> Neutral plane reference face must be planar
08:25:34  Draft001: Neutral plane reference face must be planar
08:25:48  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft001.Base missing element reference phone_cap#Draft ;#342:1;:H94e,F;:H94f,F.Face10
08:25:48  Draft:  
08:26:06  <PropertyLinks> PropertyLinks.cpp(429): phone_cap#Draft001.NeutralPlane auto change element reference phone_cap#Draft ;#296:1;:H94e,F;:H94f,F.Face5 -> Face5
08:26:06  Adding face failed on Face15. Omitted
08:26:12  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.Base missing element reference phone_cap#Cylinder003 ;#2a0:1;:H94e,F.Face12
08:26:12  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.NeutralPlane missing element reference phone_cap#Cylinder003 ;#2b7:3;:G#3c3;RFI;:H94e:b,F.Face5
08:26:12  <Exception> Neutral plane reference face must be planar
08:26:12  Draft001: Neutral plane reference face must be planar
08:26:19  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft001.Base missing element reference phone_cap#Draft ;#342:1;:H94e,F;:H94f,F.Face13
08:26:19  Draft:  
08:26:41  <PropertyLinks> PropertyLinks.cpp(449): phone_cap#Draft.NeutralPlane missing element reference phone_cap#Cylinder003 ;#2b7:3;:G#3c3;RFI;:H94e:b,F.Face5
08:26:41  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face5
08:26:41  Draft: Invalid shape name ?Face5
08:26:44  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face5
08:26:44  Draft: Invalid shape name ?Face5
08:26:49  <Exception> TopoShapeExpansion.cpp(2019): Invalid shape name ?Face5
08:26:49  Draft: Invalid shape name ?Face5
```

Also my model looks wrong. All inside cylinders should have the same radius (two of them are bigger) and the similar position (two of them placed on the wrong points).

[image]

There is my file (if it helps) 

[phone-cap-3.FCStd.zip](https://github.com/user-attachments/files/18568679/phone-cap-3.FCStd.zip)

I really excited by the function of additive and subtractive meshes. It's very cool! So I want to know how can I use it without errors.
Please, help me.

### Full version info

```shell
FreeCad 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-28:
Try to use polar array

[image]

[phone-cap-3_.zip](https://github.com/user-attachments/files/18569767/phone-cap-3_.zip)

DrMoriarty, 2025-01-28:
I avoided the problem when I deleted all Drafts and changes my cylinders to cones. (my CNC works better when I transform vertical planes into slopes of around 88-89 degrees )
With Drafts I had one more problem. When I made one Draft for all eight small cylinders, my bottom elliptical plane disappeared. So my cap became bottomless :-)
So all problems were from Drafts and all other functions works perfectly.

luzpaz, 2025-01-28:
This is a draft wb issue ?

maxwxyz, 2025-01-29:
Please add steps how to reproduce this, maybe with a simpler model

DrMoriarty, 2025-01-29:
I can not reproduce the first bug, but found another one (may be).
I've made revolution with pocket and subtractive ellipsoid, than I tried to add additive cylinder with draft and finally make polar pattern. I have no errors in console, but my model disappeared.
 
The last step looks like this:

[image]

But the previous step has full model with all details:

[image]

This is my file: 

[test.FCStd.zip](https://github.com/user-attachments/files/18593142/test.FCStd.zip)

BUT when I delete the Draft all looks nice again!

[image]

So I think that Draft logic breaks something inside.

luzpaz, 2025-01-31:
> So I think that Draft logic breaks something inside.

@Roy-043 any chance you could weigh-in ?

FEA-eng, 2026-10-25:
[This](https://github.com/FreeCAD/FreeCAD/issues/19293#issuecomment-3341537231) is still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```