FreeCAD/FreeCAD #19295: Standard Shortcut for the Addon Manager

leoheck, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

We finally have a standard shortcut for opening the Settings.  https://github.com/FreeCAD/FreeCAD/pull/15536
Can we have the same for launching the Addon Manager?

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-31:
Is Alt+T A too difficult to use???

leoheck, 2025-01-31:
@Syres916 didn't know about that. If it works than it is fine for me.