FreeCAD/FreeCAD #19297: Files Being Renamed to FCBak On Exit Of FreeCad

LHC-Canada, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am seeing strange behavior with the last three releases of the dev weekly on my Windows 10 machine.  I'm using the most current dev - 40006.  When I have a freecad model file in a sub-directory on my hard drive and I open it by opening the folder in windows, and then left click and sliding it to a shortcut on the desktop for Freecad, it opens fine, and allows me to edit it.  After saving and exiting FreeCad I am then left with nothing but "FCBak" files in the subdirectory and no FCStd files.  It's a nightmare to hunt and rename them to find out what one is the one I just edited, so I have set the backup versions to a total of 1.  Note that if I move the model file from the subdirectory ontoto the desktop and then open it from there in the same manner, this behavior does not seem to happen and I am left with the FCBak and the FCStd as expected.   

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/Qt default
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.1.0
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.24
  * VendorParts
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
Does this not happen with the stable 1.0 version?
Does this only happen with certain files and could you provide them?
Could you add precise steps how you can reliably reproduce the issue?

LHC-Canada, 2025-02-03:
I've been trying - and will continue to do so, but it's such a flaky thing that sometimes it happens and sometimes not and I can't seem to get it to repeat reliably.  One thing I can report is it has not gone away with several versions of the dev weekly builds and the reliable way to correct it, is to go up one level of window and then return to the window that is open with the model files, then it corrects itself.  ie - I will have 2 .bak files in the window and no .FcStd file when exiting from FreeCad, then when I do the back/return thing to the window I have one .bak and one .FCStd file and the file is correct from the last edit.    I understand this is not much to go on, so if I can produce it reliably with a test file I will for sure report it back here.

I have no idea about the issue with 1.0 as I am only using the weekly dev builds.  Perhaps someone else that has seen this will chime in at some point too and we can sort of narrow it down.