FreeCAD/FreeCAD #19299: Matrix inverse() and invert() inconsistency.

josegegas, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Inside FreeCAD:
>>> A = FreeCAD.Matrix()
>>> A
Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1))
>>> A.inverse()
Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1))
>>> A.invert()
>>>

In documentation (https://wiki.freecad.org/Matrix_API):
inverse( )
Description: Inverts this matrix, if possible
Returns: nothing.
invert( )
Description: Returns the inverse of this matrix, if possible
Returns: a Matrix

In C++ code (src\Base\Matrix.h):
void inverse();

In xml file (src\Base\MatrixPy.xml):
vector : Base.Vector</UserDocu>
</Documentation>
</Methode>
<Methode Name="invert" NoArgs="true">
<Documentation>
<UserDocu>invert() -> None
Compute the inverse matrix in-place, if possible.</UserDocu>
</Documentation>
</Methode>
<Methode Name="inverse" Const="true" NoArgs="true">
<Documentation><UserDocu>inverse() -> Base.Matrix

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-29:
The Wiki page is no longer updated. What do you propose should be done here?

Roy-043, 2025-01-29:
I have update the WIki page anyway.

hyarion, 2025-02-10:
@maxwxyz you can close this