FreeCAD/FreeCAD #19300: Sketcher: Line coincident to parabola apex and Y-axis causes error

AeroIllini, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

Create a sketch on the XY plane.
Create an arc of a parabola, with the apex and focus both coincident to the Y-axis. (Parabola opening up or down doesn't seem to matter.)
Create a line with points coincident to both the apex and the Y-axis.

A standard line will cause an immediate application crash.

A construction line will stop displaying the sketch and give the following error:
13:55:45  Autoconstraints cause redundancy. Removing them
13:55:45  Unhandled exception in ViewProvider::eventCallback: Access violation
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
13:55:45  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
13:55:45  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation

(Following this error, any other action in the application results in an "Access violation" error. User is forced to close the application.)

Creating a line with only one point coincident and then applying a coincident constraint to the second point does NOT cause a crash.

Here is the desired state (I created the screenshot by using the one-constraint method on the line mentioned above):
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.5
  * Curves 0.6.23
  * fasteners 0.5.24
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-29:
@AjinkyaDahale FYI

AeroIllini, 2025-01-31:
I just downloaded and tried it in weekly build 40006. The error still appears, but the error message does not include the "Unhandled Base" lines:
```
09:21:58  Autoconstraints cause redundancy. Removing them
09:21:58  Unhandled exception in ViewProvider::eventCallback: Access violation
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
```

Both a standard line and a construction line caused a crash, which is different than the stable release (in the stable release, the construction line didn't close the window, but it also didn't allow any further work in the program until it was restarted, which I guess is a type of crash).

I did some more poking around, and it seems to be related to the line being **parallel** to one of the Sketch axes, not necessarily **on** an axis. All of the following are in the weekly 40006 build.

The horizontal axis also causes a crash, same as above.
**CRASH:**
[image]

Performing the procedure above on a set of angled lines does NOT cause a crash.
**NO CRASH:**
[image]

But setting the center line to vertical (or horizontal) causes the same crashing behavior.
**CRASH:**
[image]

maxwxyz, 2025-02-03:
@AjinkyaDahale @FlachyJoe FYI

AjinkyaDahale, 2025-03-14:
Cannot reproduce on
```
OS: Ubuntu Core 22 (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.40717 (Git) Snap 1338
Build date: 2025/03/13 13:32:18
Build type: Release
Branch: main
Hash: a375752aa4ea4d416f757299c716820546fd41cc
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * OpenTheme 2024.9.1
```
There are however some redundant autoconstraints that are automatically removed.

luzpaz, 2025-07-21:
@AeroIllini can you reproduce on 1.1dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

AeroIllini, 2025-07-27:
@luzpaz Still throws an error, and still crashes.

I downloaded build 42721.

[image]

```
21:01:46  Autoconstraints cause redundancy. Removing them
21:01:46  Unhandled exception in ViewProvider::eventCallback: Access violation
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
21:01:46  Autoconstraints cause redundancy. Removing them
21:01:46  Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!
```

What's weird is the first time I did it, it threw the error above, but it created two vertical lines after I placed the second coincident point (and neither of the lines had a second coincident point). The second time I tried it from a fresh start, and every subsequent time since, the program crashed immediately after placing the vertical line.

AeroIllini, 2025-07-28:
Downloaded the 7-26-2025 weekly.

It works!

[image]

No errors, no crashes. Just a standard warning about redundant autoconstraints. It looks like you squashed it.