FreeCAD/FreeCAD #19304: Trying to pad a ShapeString makes everything disappear when I try to pad it

Equanico, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I try to pad a shapestring it causes the canvas to blank out as if all my objects disappear. If the shapestring isn't touching anything only the shapestring disappears, but if it's touching the body it makes the body disappear with it. I'm not sure if I'm doing something wrong or if It's a bug.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-01-29:
In order to assist, at the very least please specify the font and the word you are trying to pad so we can see which letter is the root cause. On Windows, have you tried the usually reliable Arial font?

Roy-043, 2025-01-29:
This may just be a contiguous solid issue.
https://wiki.freecad.org/Feature_editing#What_is_a_contiguous_solid?

Roy-043, 2025-01-29:
Please post your FCStd file.

closed 2025-01-29 by Equanico