FreeCAD/FreeCAD #19305: [Draft/BIM] Problem with transformations along the main axes when using the XYZ keys

kaiwas, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If I made small movements with the mouse before pressing X, the data from the movements is not reset. That is, despite the fact that I assumed movement along X, I also get movement along I.
In this case, I cannot switch to the I axis to change the value. TAB and I do not work.

https://github.com/user-attachments/assets/5691fb2c-accb-4380-a5ec-683c23716948

I will repeat for the hundredth time that capturing data from mouse movement when it is not required is a terrible problem for freecads, but it can be mitigated if you reset the data at the right moment.

For example, moving with the XYZ keys in blender. At the beginning, I move the object randomly and the values ​​of all axes are accepted. But after pressing X, I move only along the X axis.

https://github.com/user-attachments/assets/1d050427-4438-4cf2-8af2-55119acdc4b9

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2 1.0.0
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.1.0
  * freecad.gears 1.3.0
  * sheetmetal 0.7.3
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-29:
Confirmed. You have to move the mouse in the 3D view for the coords to be updated correctly.

kaiwas, 2025-01-29:
> You have to move the mouse in the 3D view

Yes. Thank you!
It seems that the problem strongly depends on the order of actions. But it seems to me that the tool should work the same in any order of actions. It is the different approaches to "intercepting" values ​​from mouse movements, or when entering values ​​in fields - that cause misunderstanding.

PR #19343 (2025-02-04): Draft: fix three issues with DraftGui.py
Fixes #19305.
Fixes #19324.

Issues above caused by #19179.

Also removes automask code as it did not work properly in several cases. For example if the working plane was set to Front and the Global option was checked only the X coordinate could be entered.
commit 6b76430c (2025-02-04): Draft: fix three issues with DraftGui.py (#19343)

Fixes #19305.
Fixes #19324.

Issues above caused by #19179.

Also removes automask code as it did not work properly in several cases. For example if the working plane was set to Front and the Global option was checked only the X coordinate could be entered.
closed 2025-02-04 by Roy-043