FreeCAD/FreeCAD #19308: PartDesign: Suppressed feature does not

leoheck, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This "suppressed" item is available in the context menu.

When an item is suppressed, it shows no difference in the Objects Tree. 

For instance, this is a suppressed Feature.
Can you tell which feature is suppressed here?
[image]

ALSO... While playing with this suppressed feature.
I got this SEGFAULT at my face.
I am suppressing a feature that is broken with Error. This may help something.
If the Feature is suppressed, why is it trying to compute it? Shouldn't it be ignored?

```
<PartDesign> FeatureDressUp.cpp(221): Ground_Sensor#Fillet031: skip invalid shape '' with type Solid
<Exception> Fillet not possible on selected shapes
<PartDesign> Feature.cpp(98): Failed to recompute suppressed feature Ground_Sensor#Fillet031
Sketch351: AttachEngine3D: subshape not found Chamfer024.?Face419
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7862afc45320]
#1  0x786290b51fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCAFLeKmP/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x786290b63b21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCAFLeKmP/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x786290b64d35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCAFLeKmP/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x7862a09cf1ae in Measure::Measurement::length() const from /tmp/.mount_FreeCAFLeKmP/usr/lib/Measure.so+0x15e
#5  0x78627f4ad45e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAFLeKmP/usr/lib/MeasureGui.so+0x31e
#6  0x78627f4adcbd in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAFLeKmP/usr/lib/MeasureGui.so+0x5d
#7  /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7862b02eb6f2]
#8  0x7862b02ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x7862b02e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x7862b11703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x7862b2f74dd9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libFreeCADGui.so+0xc9
#12  0x7862b02b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x7862b0309f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5(+0x30a75c) [0x7862b030a75c]
#15  /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7862ae910ad7]
#16  /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7862ae913ac7]
#17  /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7862ae9140ce]
#18  0x7862b030aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x7862b02b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x7862b02bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x7862b2ed4d3a in Gui::Application::runApplication() from /tmp/.mount_FreeCAFLeKmP/usr/bin/../lib/libFreeCADGui.so+0x8aa
#22  /tmp/.mount_FreeCAFLeKmP/usr/bin/freecad(+0x78f6) [0x5feb9a7058f6]
#23  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x7862afc2a1ca]
#24  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7862afc2a28b]
#25  /tmp/.mount_FreeCAFLeKmP/usr/bin/freecad(+0x7c7e) [0x5feb9a705c7e]
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.6.13
  * Curves 0.6.53
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.35
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.4.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-01-30:
Today I could see a different behavior. Some suppressed items are shown with a strikethrough.
But `Chamfer025`, a problematic feature, is suppressed but does not have this strikethrough.
[image]

FlachyJoe, 2025-01-31:
Hi,
The Suppressed features are computed to keep the references up to date.
If you refer an sub-element of a suppressed feature on a feature below this one has to know the element.

I introduced a Frozen state to block the recompute of an object but it currently doesn't work as I like.

@maxwxyz you can assign me but I don't know when I'll can take a look.

leoheck, 2025-01-31:
Hi @FlachyJoe. So suppressed features are computed but not displayed. And since `Chamfer025` is broken, the suppression is pending... Is this what you meant to say, right?

I still think that the strikethrough should be displayed even if the feature is not computed since it is already set by the user as suppressed. It should apply the strikethrough and also the broken symbol.

What is the workflow to delete a broken feature like this one? I want to delete it since I cannot guarantee that I know how many items were there before.

Should I update the references to it, like changing the base of Fillet032 and any other direct reference, and then I can proceed with deleting the feature? 

Because of how the TopoNaming is implemented it is quite hard to fix Chamfers and Fillets that are problematic nowadays. In the past, broken items would keep being listed. So the user could explore which of each should be changed or which one should be removed.

Now, references on those old items disappear from the list. For instance, I have a chamfer of 10 edges. Then after changing something before this feature in the design, I will see less than 10 edges in the list, sometimes even a list without items. Having the same original number of items in the list would help fix edges that do not exist anymore or edges that were renamed. So the fix would be easy since the user could work to restore the number of edges chamfered/filleted. If the number of chamfered/filleted is updated and kept to the same amount, the next features of the body after the chamfer/fillet will continue working straight without too much work.

FlachyJoe, 2025-02-07:
@leoheck can you share a failing file please?

luzpaz, 2025-05-05:
@leoheck 👆

closed 2025-05-06 by leoheck