FreeCAD/FreeCAD #19310: AddonManager: should identify Snap packaging and use `freecad.pip` instead of using python pip installed on the system

luzpaz, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Issue derives out of
* https://github.com/FreeCAD/FreeCAD-snap/issues/152

Where we learned that in a Snap build of FreeCAD, the Addon Manager when trying to install the 3rd party 'InventorLoader' addon which has `ezdxf` as a python dependency, tried to run a python3.10 binary that wasn't on the users' system to install the dependency. This is problematic because of the sandboxed nature of Snap packaging + the version collisions that would ensue. 

The remedy is for the Addon Manager to recognize:
1. that it's being run within a Snap package
2. FreeCAD Snaps come with a `FreeCAD.pip` CLI function that can be used to install 3rd party python libraries. 
3. Addon Manager should leverage `FreeCAD.pip` over it's current behavior

### Full version info

```shell
1.0 + 1.1dev
```

### Subproject(s) affected?

Addon Manager

### Anything else?

Thanks to @tmarkettos for reporting and @furgo16 for diagnosing.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-29:
Here's how to test if we're currently in a Snap package:

```
import os

snap_package = os.getenv("SNAP_REVISION")

if snap_package:
    print(f"Snap Revision: {snap_revision}")
else:
    print("Snap Revision environment variable is not set.")
```

Looks like we need to add a conditional in 
https://github.com/FreeCAD/FreeCAD/blob/1e05c59ce897198ccfdde300599a908d48d03c36/src/Mod/AddonManager/addonmanager_python_deps_gui.py#L106-L108

to make an exception for Snap packages to use `freecad.pip`

Note: 
```shell
$ which freecad.pip                                                                                                                  
/var/lib/snapd/snap/bin/freecad.pip
```

Syres916, 2025-01-29:
Another report https://forum.freecad.org/viewtopic.php?t=94270

luzpaz, 2025-01-31:
@chennes any chance you could implement this?

chennes, 2025-01-31:
Is `freecad.pip` located in a known location?

luzpaz, 2025-01-31:
AFAICT, yes. `/var/lib/snapd/snap/bin/freecad.pip`

chennes, 2025-01-31:
Is that in `PATH`?

luzpaz, 2025-01-31:
```shell
echo $PATH                                                                                                                                                                                                                                                                                                                                                                  ✔ /home/foobar/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin
```

Yes, it's at the end of PATH

luzpaz, 2025-01-31:
> Can a snap package be installed in different location? 
>
> No, a Snap package itself cannot be installed in a different location other than the designated `/snap` directory on your system, but a Snap application's user data can be stored in a different location depending on the user's home directory, allowing some flexibility in where the application stores its files within the user's space; however, the core Snap package files will always reside within the `/snap` directory

closed 2025-02-04 by chennes