FreeCAD/FreeCAD #19319: Assembly importing 2 times a sub assembly causes dead lock or infinite loop

FredLaCasemate, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I created a file modeling a simple hinge. In a second file I create a simple panel and then an assembly in which I add the panel twice and the external hinge twice. I managed to make the assembly work with the two panels and a hinge. If I add the second hinge it fails to join it correctly to the assembly. On one side I can fix it to one of the panels, but if I join the second half of the hinge to the other panel FreeCad crashes. I tried both with a fixed joint and with a parallel joint.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.7.5
```

### Subproject(s) affected?

Assembly

### Anything else?

[Charnière V2.zip](https://github.com/user-attachments/files/18594723/Charniere.V2.zip)
(Note : this file use SheetMetal workshop)

[TestCharnière.zip](https://github.com/user-attachments/files/18594727/TestCharniere.zip)

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-30:
Does it work in the stable version? There is no crash or error on my side, can you add a screencast or steps how to reproduce?
```
OS: macOS 13.7.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45

```

FredLaCasemate, 2025-01-31:
Hi,
The first step on the file is to create a new fixed join between the left panel face (Corps002.Face6) and the down side of the hinge (Body005.Face3) with the correct offsets.

[image]

After this creation, try to move the panel. Then freecad enters in an infinite loop and the only mean to stop is to kill the process.

[image]

I tried that as well in the stable version (on Linux and on Windows) and in the last dev vesion (on Linux)

Best Regards
Fred

maxwxyz, 2025-02-03:
@PaddleStroke is this a known issue that you cannot import a sub-assembly twice?

PaddleStroke, 2025-02-03:
No I didn't knew. it should work.

PaddleStroke, 2025-07-23:
ok so the assembly solves when we create the joint. It's the dragging that is failing.

PR #22685 (2025-07-24): Assembly: Fix infinite loop with subassembly. Fix #19319
Fix #19319

The issue is that getUpstreamMovingPart was entering an infinite loop where the same joint would be selected over and over. I'm not sure why this happened only in the case of sub-assemblies as it sounds like something that should happen quite often.

Anyway it's fixed by making sure a joint is not traversed twice.

Also removes old & unused getUpstreamParts
closed 2025-07-24 by kadet1090