FreeCAD/FreeCAD #19320: Keyboard shortcuts when using Shift are assigned incorrectly

00001000bit, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

_Note: Identified on FC 1.0.0.39109 (both mac and win).
But also tested on 1.1.0dev 40006 to confirm it is still relevant - same behavior._

Attempted to change the shortcuts for the standard views (0-6) by requiring the use of the Shift key in combination with the number.

When attempting to type into the "New shortcut" box to detect the new key, it assigns it to the Shift key in combination with the "shifted" alternate version of the key, and not the standard version (for example: as "Shift+%" rather than "Shift+5" when using a US keyboard).

When attempting to USE the shortcut, it doesn't cannot be invoked. (Pressing Shift and 5 - nothing happens)

However, if I manually edit user.cfg for the appropriate view (in this case), changing the stored value:
<FCText Name="Std_ViewBottom">Shift+%</FCText>
to
<FCText Name="Std_ViewBottom">Shift+5</FCText>
The correct key combination will be detected on the next run. So, it appears to be an issue only in attempting to assign the shortcut, not in using it.

### Full version info

```shell
OS: macOS 14.7.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=94150&sid=07c6ad713447281207e5259811809fe3

(Link to forum, but no real discussion at this time - 1 week later.)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-01-30:
Which navigation style are you using, some use the Shift key for changing navigation.
Do other shortcuts with Shift work (e.g. Shift+Left)?

00001000bit, 2025-01-30:
I'm sing Touchpad as navigation style, but shift is working everywhere else as I expect.

Even USING existing Shift key shortcuts works fine. 

It's just that I can't assign NEW ones properly when there is an alternate glyph on the key (like with the numbers) as it assigns them as Shift+(alternate glyph) instead of Shift+(actual key). Letters work fine, it seems to be the keys that have different glyphs on the keys rather than just lower/capital.

So, if I try to assign something to Shift-L it assigns and works properl
y.
But, if I try to assign something to Shift-(number) it assigns it as Shift-(whatever alternate glyph is on the key)

So, when I try to use those ones, they don't work.
If I manually edit the user.cfg to make it Shift-(number) - then using the shortcut works without issue.

The same thing happens on non-number keys that have multiple glyps, such as semicolon (colon).
If I try to assign something to Shift+(semicolon), it gets saved as Shift+(colon) and cannot be used. Again, if I manually edit the user.cfg file, the shortcut works as it should.

It seems the problem is during the assignment, it's reading the shift key as a modifier, but also attaching it to the modified glyph, which sort of doubles up the modifier making it impossible to type. (For whatever reason, letters are not affected in this manner.)