FreeCAD/FreeCAD #19325: Link objects scale property doesn't work when transfrom tool used

ErtugrulEge, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you use the transform tool after setting scale property , link objects do not maintain their scaled state and return to their previous state. In version 1.0 scale property working well.

Forum topic: https://forum.freecad.org/viewtopic.php?t=94347

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
@kadet1090 is this an issue with the transform tool or links?

mnesarco, 2025-03-09:
There is another maybe related issue, scale alters object Placement, so if the link is attached to a Face, changing scale moves the object away from the Face because Z is scaled also.

1. Object attached to Face, scale = 1 

[image]

2. Changing Scale = 1.5 moves the object away of the support Face

[image]

OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Mnesarco_Utils 0.2.15
  * Marz 0.1.15
  * FreeCAD_Svg 1.0.0.dev3
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * fcapi

luzpaz, 2025-09-01:
@ErtugrulEge do you mind uploading an example file for testing convenience ?

closed 2025-09-01 by maxwxyz