FreeCAD/FreeCAD #19327: BIM: Error when exporting to .ifcJSON

semhustej, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- start a new file
- create a wall
- select the wall
- File -> Export
- select .ifcJSON
- select `Save`
- Error shows up, no file is saved

`Error: Unable to locate ifc2json5a module. Aborting.`

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was discovered by a user with FC 1.0 version in this forum thread: https://forum.freecad.org/viewtopic.php?t=94307

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-31:
Sounds like a regression

semhustej, 2025-01-31:
> Sounds like a regression

I did not test with versions previous to 1.0 so I don't know if it ever worked.

Syres916, 2025-01-31:
If the user has even installed `ifcopenshell 0.8.0`, there is no module called `ifcjson` or `ifc2json`, the only place I can find `ifcjson5a` is at https://github.com/buildingsmart-community/ifcJSON but I can't find a proper `pip` install for it so I'm struggling to see how this has ever worked.

yorikvanhavre, 2025-02-04:
IIRC it needs that module installed manually... Indeed this is far from user-friendly 😅 
Not sure how to go to solve this... Including ifcjson in our source code, although technically possible (MIT license), is probably not what we want

yorikvanhavre, 2025-02-04:
Bonsai offers json export. It would be interesting to see how they go about it, and maybe we just do the same...

furgo16, 2025-05-22:
Also, there are three different locations for this module:

1. https://github.com/buildingsmart-community/ifcJSON/tree/main/file_converters/ifcjson
2. https://github.com/IFCJSON-Team/IFC2JSON_python/tree/master/file_converters/ifcjson
3. https://github.com/Moult/IFC2JSON_python/tree/feature-ios-v0.8.0/file_converters/ifcjson

Location 3 is the one Bonsai BIM uses. It's a patched version of location 2 adapted to ifcopenshell 0.8.0. Bonsai uses it since the PR to adapt upstream to ifcpenshell 0.8.0 (https://github.com/IFCJSON-Team/IFC2JSON_python/pull/3/files) has not been merged yet.

That would indicate that location 2 is the upstream, but I don't know which relationship it bears to location 1.

It's a small module and the code hasn't changed in 5 years. Another alternative to the way Bonsai BIM do it in their makefile could be to simply include it in https://github.com/FreeCAD/FreeCAD/tree/main/src/3rdParty

yorikvanhavre, 2025-05-23:
> @chennes Do you think it would be somehow possible to include this in our packaging? Otherwise we would have to implement our own IFC dumper it seems.

Responding for him here, the addon manager can only install 2 things: 1) addons from our FreeCAD addons repo, or 2) Python packages from pip. ifc2json is not in pip AFAICS, but maybe an easy solution would be to create a FreeCAD addon with it?

yorikvanhavre, 2025-05-26:
Basically create (or fork) a new ifc2jason repo and add it as a gitmodule on https://github.com/FreeCAD/FreeCAD-addons :)

All addons base folders are added to the python path in FreeCAD. So let's say our addon is called "ifc2json-freecad", when installing the addon, a ifc2json-freecad folder is created in the /Mod folder, and the contents of the addon are placed there. The ifc2json-freecad folder is added to the FreeCAD python path, so anything inside is now importable in python.

For ex, if it contains a ifc2json.py, now we can just `import ifc2json`

furgo16, 2025-05-26:
Sounds good. If we were to "package" the module as an add-on, is there a mechanism to include add-ons such as this one at build time? Or would the user need to install the add-on manually after the FreeCAD installation?

yorikvanhavre, 2025-05-26:
Some addons could be shipped with FreeCAD (I don't think any is, at the moment, but it has been done by ondsel). But in that case, it's better to just include it as a built-in mod in the FreeCAD source, or inside the BIM WB itself.

But in both cases, that triggers license issues. We must be careful about the license, see that it's all properly licensed and formatted, etc.

Also another problem is that we become responsible for maintaining it, keeping in sync with the upstream changes, etc... It's more maintenance work for us.

I any case, the BIM WB still depends on a couple of external parts, ifcopenshell being of course the main one. So we can't escape some annoyances just yet. And since json ifc seems not very much used yet, I guess this is not a super urgent issue...

Roy-043, 2026-09-04:
For such a simple module just downloading the files would also be possible. The BIM Library code has all the required tools.