FreeCAD/FreeCAD #19328: Start page toggle not in preferences (1.0 and weekly build 40006)

AnotherCurtis, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I could not figure out how to access the start page to toggle it back on at startup. I went straight the general tab of preferences thinking I would see a checkbox for "Don't show start page at startup" checked.

I went to the [wiki](https://wiki.freecad.org/Start_Preferences/en) which still lists it as a workbench with preferences.

I then searched the web for a bit. Found someone else on the forum with the same problem. Someone had answered it is in the Help menu. This was not intuitive for me. It has more to do with dealing with files than being helpful. I had searched the File menu, the View menu, and the Windows menu. I knew that Help would take me to the wiki so I never looked in the Help menu, instead heading directly to the wiki.

I asked a few people I know, if they knew how to toggle the start page and they all headed into preferences looking for it. 

I realize that if it was only a checkbox in preferences (without a menu item) the page could only be accessed, once closed, by restarting the application. It's not a great solution. Moving the menu elsewhere might be more intuitive but where to be the most obvious for the majority is a difficult question.

I [posted to the forum](https://forum.freecad.org/viewtopic.php?p=807744&sid=1aa312b3a4b02d773ba0fd3adfab7f2d#p807744) about the whole thing and was told to maybe bring it over here for further discussion.

I'm not complaining. Mostly just food for thought. Thank you to everyone responsible for FreeCAD. It is a great thing.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.1.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
@Roy-043 any chance to update the wiki? Who could I ping?

closed 2025-02-03 by maxwxyz

Roy-043, 2025-02-04:
I have added a banner to the mentioned preference page.