FreeCAD/FreeCAD #19329: [TechDraw] - Crashing and sluggishness when inserting views of a simple part

vancampbeans, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am experiencing issues in TechDraw when trying to create views ("Insert View" button) with a **specific part**. It takes TechDraw an absurd amount of time to generate the view (regardless of if you choose front, side, isometric, etc). After the view has been generated, FreeCAD also becomes slightly sluggish (depending on the computer specs). I tested this on three separate computers:

1. Thinkpad X220 Intel i5 Sandy Bridge 16GB RAM with Debian 12
2. Thinkpad T420 Intel i7 Ivy Bridge 16GB RAM with Debian 12
3. Desktop Computer Intel i7-6700 with Windows 10

This part is part of a larger assembly I am working on which generates the views in TechDraw without any issues if the problem part is removed. As soon as the problem part is reintroduced, it is even more sluggish and caused the RAM on my X220 to surpass 16GB resulting in a crash of the entire system. So I was able to isolate these issues to be caused by this single part. I have also re-drawn it many times and it is still has the same issues.

The part itself is quite simple. Just a sketch that is revolved around Z axis. Another two sketches and then holes are pocketed through the part. Finally, two radial pattern operations on the two pockets to propagate the holes around the cylindrical part.

**Steps to recreate the problem:**
1. Open FreeCAD.
2. Create a new, completely blank empty file.
3. Run "testpart2.FCMacro" macro (attached in .txt).
3. Move to TechDraw Workbench.
4. Insert Default Page.
5. Select "Body" from the workspace tree, then select "Insert View".
6. Observe the time taken to generate the view.

**NOTES:**
1. After selecting "Body" and "Insert View", if you select "Cancel" in the Insert View window before selecting "Apply" and or "OK" causes FreeCAD to crash.
2. Pressing "Apply" and "OK" same as NOTE 1 before the view preview is generated will not crash FreeCAD. However, if you then try to delete the TechDraw page, or close FreeCAD, it will crash FreeCAD.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods:
* quetzal-master 1.0.1
```

### Subproject(s) affected?

TechDraw

### Anything else?

[testpart2.txt](https://github.com/user-attachments/files/18610022/testpart2.txt)

Forum Post: https://forum.freecad.org/viewtopic.php?p=808058#p808058

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-01-31:
The delay is in OCC's HLRAlgo (hidden line removal) processing.  If you change the projection direction slightly from (0, -1, 0) to (0.001, -1, 0) it changes the angles enough that OCC doesn't struggle.

original direction
11:17:13  View is finding hidden lines
11:17:13  GO::projectShape()
11:22:45  GO::projectShape results
11:22:45  GO::projectShape - hlrAlgo: 297869    ~5   minutes
11:22:45  GO::projectShape - hlrToShape: 1350   ~1   second
11:22:45  GO::projectShape - makeTD: 32797      ~32  seconds
11:22:45  GO::projectShape - total: 332017      ~5.5 minutes
11:22:45  View has finished finding hidden lines
11:22:45  View is extracting faces
11:22:46  View has finished extracting faces

direction altered to (0.001, -1, 0)
12:36:00  View is finding hidden lines
12:36:00  GO::projectShape()
12:36:02  GO::projectShape results
12:36:02  GO::projectShape - hlrAlgo: 1075      ~1 second
12:36:02  GO::projectShape - hlrToShape: 39     <1 second
12:36:02  GO::projectShape - makeTD: 987        <1 second
12:36:02  GO::projectShape - total: 2102        ~2 seconds
12:36:02  View has finished finding hidden lines
12:36:02  View is extracting faces
12:36:02  View has finished extracting faces

BTW, I believe your crash is fixed by PR #19227.

maxwxyz, 2025-02-03:
is the entire issue closed with #19227?
If not, what is the action required to closing this?

WandererFan, 2025-02-04:
The slowness of the HLR calls is still an issue, even without a crash.

tringenbach, 2025-03-25:
Here's what MacOS's Activity Monitor's "Sample" button shows when I follow the instructions listed here.

I thought maybe I did it wrong, because it didn't lock up or show a shape, so I ended up having three going at once before my laptop fans spun up and I realized that it was actually working.

I was on hash 723a25c57d526b65055bf7dd29d3ab58ef8b28c2 from March 16th, plus some unrelated local changes I was working on. (Sorry it's not clean, but this should still give us an idea of where it's spending its time.)

 [Sample of FreeCAD-hlr-bug.txt](https://github.com/user-attachments/files/19457777/Sample.of.FreeCAD-hlr-bug.txt)

oursland, 2025-03-26:
I ran the script and performed a Techdraw Insert View within the Xcode Instruments Time Profiler.  This took 8 minutes to generate the 2D image.  The results differ from the previously reported claims.

The time was spent primarily in the functions:

| Func | Time |
|--|--|
| HLRBrep_Hider::Hide() | 7.38 min |
| HLRBRep_Data::Classify() | 4.44 min |
| BRepClass_FClassifier::Perform() | 4.27 min |
| BRepClass_FClass2dOfFClassifier::Compare() | 3.10 min |
| BrepClass_Intersector::Perform() | 3.09 min | 
| Extrema_ExtPC2d::Extrema_ExtPC2d() | 1.32 min |
| Extrema_ExtPC2d::Perform() | 1.30 min |
| Extrema_ExtPC2d::IntervalPerform() | 28.31 s |

[image]

tringenbach, 2025-03-26:
Here's some the stats from the log I posted.

**Call Stack**
It took 2109 samples. Anything that says "2109" was running for the whole couple of seconds that it sampled.

`HLRBRep_Hider::Hide` then is the first one it caught not running the whole time.  I think this aligns pretty well with what you posted @oursland .

```
    2109 Thread_11555728: Thread (pooled)
    + 2109 thread_start  (in libsystem_pthread.dylib) + 15  [0x7ff8087afbef]
    +   2109 _pthread_start  (in libsystem_pthread.dylib) + 99  [0x7ff8087b4253]
    +     2109 QThreadPrivate::start(void*)  (in libQt6Core.6.7.3.dylib) + 493  [0x1079598dd]
    +       2109 QThreadPoolThread::run()  (in libQt6Core.6.7.3.dylib) + 452  [0x107960684]
    +         2109 QtConcurrent::RunFunctionTaskBase<void>::run()  (in TechDraw.so) + 97  [0x198c2b2c1]
    +           2109 QtConcurrent::StoredFunctionCall<TechDraw::DrawViewPart::buildGeometryObject(TopoDS_Shape&, gp_Ax2 const&)::$_1>::runFunctor()  (in TechDraw.so) + 163  [0x198c45a03]
    +             2109 TechDraw::GeometryObject::projectShape(TopoDS_Shape const&, gp_Ax2 const&)  (in TechDraw.so) + 270  [0x198cd6c7e]
    +               2109 HLRBRep_InternalAlgo::Hide()  (in libTKHLR.7.8.1.dylib) + 219  [0x11f5d213b]
    +                 2109 HLRBRep_InternalAlgo::Hide(int)  (in libTKHLR.7.8.1.dylib) + 252  [0x11f5d1fdc]
    +                   2109 HLRBRep_InternalAlgo::HideSelected(int, bool)  (in libTKHLR.7.8.1.dylib) + 3566  [0x11f5d32ae]
    +                     714 HLRBRep_Hider::Hide(int, NCollection_DataMap<TopoDS_Shape, BRepTopAdaptor_Tool, TopTools_ShapeMapHasher>&)  (in libTKHLR.7.8.1.dylib) + 4251  [0x11f5bf4ab]
    +                     ! 713 HLRBRep_Data::Classify(int, HLRBRep_EdgeData const&, bool, int&, double)  (in libTKHLR.7.8.1.dylib) + 3478  [0x11f5b7a26]
    +                     ! : 695 BRepTopAdaptor_FClass2d::Perform(gp_Pnt2d const&, bool) const  (in libTKTopAlgo.7.8.1.dylib) + 1084  [0x11f9c379c]
    +                     ! : | 693 BRepClass_FaceClassifier::Perform(TopoDS_Face const&, gp_Pnt2d const&, double, bool, double)  (in libTKTopAlgo.7.8.1.dylib) + 112  [0x11f943800]
    +                     ! : | + 418 BRepClass_FClassifier::Perform(BRepClass_FaceExplorer&, gp_Pnt2d const&, double)  (in libTKTopAlgo.7.8.1.dylib) + 630  [0x11f945bc6]
    +                     ! : | + ! 418 BRepClass_FClass2dOfFClassifier::Compare(BRepClass_Edge const&, TopAbs_Orientation)  (in libTKTopAlgo.7.8.1.dylib) + 88  [0x11f945348]
    +                     ! : | + !   171 BRepClass_Intersector::Perform(gp_Lin2d const&, double, double, BRepClass_Edge const&)  (in libTKTopAlgo.7.8.1.dylib) + 518  [0x11f946286]
    +                     ! : | + !   : 170 Extrema_ExtPC2d::Extrema_ExtPC2d(gp_Pnt2d const&, Adaptor2d_Curve2d const&, double)  (in libTKGeomBase.7.8.1.dylib) + 289  [0x12017d611]
    +                     ! : | + !   : | 58 Extrema_ExtPC2d::Perform(gp_Pnt2d const&)  (in libTKGeomBase.7.8.1.dylib) + 6982  [0x12017ba86]
    +                     ! : | + !   : | + 54 Extrema_ExtPC2d::IntervalPerform(gp_Pnt2d const&)  (in libTKGeomBase.7.8.1.dylib) + 101  [0x12017ce65]
    +                     ! : | + !   : | + ! 51 Extrema_EPCOfExtPC2d::Perform(gp_Pnt2d const&)  (in libTKGeomBase.7.8.1.dylib) + 102  [0x120162f96]
    +                     ! : | + !   : | + ! : 26 math_FunctionRoots::math_FunctionRoots(math_FunctionWithDerivative&, double, double, int, double, double, double, double)  (in libTKMath.7.8.1.dylib) + 881  [0x11f3213e1]
```

**Sort by top of stack**
Ignoring the ones that are just other threads waiting, `BSplCLib::Bohm` wins this one. `Hide` never shows up here since the sampler never took a sample where it was running code inside of `Hide`, it only caught it running functions called by `Hide`.
```
Sort by top of stack, same collapsed (when >= 5):
        __psynch_cvwait  (in libsystem_kernel.dylib)        31635
        mach_msg2_trap  (in libsystem_kernel.dylib)        4196
        __select  (in libsystem_kernel.dylib)        2109
        __workq_kernreturn  (in libsystem_kernel.dylib)        2109
        BSplCLib::Bohm(double, int, int, double&, int, double&)  (in libTKMath.7.8.1.dylib)        492
        PLib::NoDerivativeEvalPolynomial(double, int, int, int, double&, double&)  (in libTKMath.7.8.1.dylib)        362
        TopLoc_SListOfItemLocation::Assign(TopLoc_SListOfItemLocation const&)  (in libTKMath.7.8.1.dylib)        329
        _nanov2_free  (in libsystem_malloc.dylib)        320
        nanov2_calloc_type  (in libsystem_malloc.dylib)        319
        TopLoc_Location::Multiplied(TopLoc_Location const&) const  (in libTKMath.7.8.1.dylib)        190
        TopLoc_SListOfItemLocation::TopLoc_SListOfItemLocation(TopLoc_ItemLocation const&, TopLoc_SListOfItemLocation const&)  (in libTKMath.7.8.1.dylib)        180
        NCollection_BaseSequence::Find(int) const  (in libTKernel.7.8.1.dylib)        127
        CSLib_Class2d::InternalSiDans(double, double) const  (in libTKMath.7.8.1.dylib)        118
        BSplCLib_Cache::D0(double const&, gp_Pnt2d&) const  (in libTKMath.7.8.1.dylib)        116
        math_FunctionRoots::math_FunctionRoots(math_FunctionWithDerivative&, double, double, int, double, double, double, double)  (in libTKMath.7.8.1.dylib)        115
        TopLoc_ItemLocation::TopLoc_ItemLocation(opencascade::handle<TopLoc_Datum3D> const&, int)  (in libTKMath.7.8.1.dylib)        104
[... cut ...]
        39       BSplCLib::BuildCache(double, double, bool, int, int, NCollection_Array1<double> const&, NCollection_Array1<gp_Pnt2d> const&, NCollection_Array1<double> const*, NCollection_Array2<double>&)  (in libTKMath.7.8.1.dylib) + 201  [0x11f37c8e9]
        39       BSplCLib_Cache::BuildCache(double const&, NCollection_Array1<double> const&, NCollection_Array1<gp_Pnt2d> const&, NCollection_Array1<double> const*)  (in libTKMath.7.8.1.dylib) + 317  [0x11f38932d]
        38       nanov2_calloc_type  (in libsystem_malloc.dylib) + 625  [0x7ff8085e92f7]
```