FreeCAD/FreeCAD #19330: Freecad Daily Builds - 'freecad-daily-python3' has no installation candidate (arm64)

eleknader, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to install freecad-daily from ppa:freecad-maintainers/freecad-daily

System is Ubuntu Server 24.04.01 on Virtualbox, machine Mac OSX 15 on Apple M5.
Clean install of Ubuntu with KDE Plasma desktop.

Install fails with error: freecad-daily-python3 is not installable

sudo apt-get install freecad-daily-python3
results
E: Package 'freecad-daily-python3' has no installation candidate

BR

Tapio

### Full version info

```shell
Freecad daily ppa
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-01-31:
The package builds look ok at a first glance: https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily/+packages

From your report, it seems you're trying to install the `freecad-daily-python3` directly. Could you please run the instructions from the wiki at https://wiki.freecad.org/Installing_on_Linux#Development_PPA_(Daily) and post the full log?

```
sudo add-apt-repository ppa:freecad-maintainers/freecad-daily
sudo apt-get update
sudo apt-get install freecad-daily
```

furgo16, 2025-01-31:
Oh, I see, you're on `arm64`. As far as I can tell, there are no arm builds available.

gramss, 2025-10-25:
Hello there,

I just wanted to check if this is on a roadmap or there might be a work around?
It seems to me that a standard python3 install won't cut it.

Thanks!