FreeCAD/FreeCAD #19331: When moving away from the origin, the snap endpoint shows the wrong location on end points (visual only)

ErtugrulEge, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, I have some point  coordinates (X,Y) shows boundaries of the land.I have to draw lines with this coordinates but those coordinates are far from origin and showing some gaps between continous lines and, endpoint snap showing inccorrect places just visually because  when you move lines to origin there is no gap and in fact everything is fine. There is a visualisation problem when you far away origin. 

I did the same thing in LibreCAD and It's working fine far away from origin.

File and video in forum topic: https://forum.freecad.org/viewtopic.php?t=94384

 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-01-31:
This is an issue that is not unique to the Draft WB. It is Core issue, and is also related to OCC and its limitations.

Roy-043, 2025-01-31:
Related: #19326.

stefano2734, 2026-09-12:
Yes, occt 7.8.1 is quite old.
It should be tested again with occt 7.9.3,  8.0.1 or later.

ErtugrulEge, 2026-09-12:
Seems situation is same to me 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260908 (Git shallow)
Build date: 2026/09/08 23:17:32
Build type: Release
Branch: (HEAD detached at 306259d7a)
Hash: 306259d7aa3e7f3b40a22232eb60973deb39ca20
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR) [ OS: Turkish/Türkiye (tr_TR) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.06/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curves workbench 0.6.70
```

stefano2734, 2026-09-12:
> Seems situation is same to me
>
> ```
> OS: Windows 11 build 26200
> Architecture: x86_64
> Version: 26.3.0dev.20260908 (Git shallow)
> Build date: 2026/09/08 23:17:32
> Build type: Release
> Branch: (HEAD [...]

I see here occt 7.8.1, but 7.9.3 (last of 7.x) and 8.0.1 are available.
So no progress is here, when all test with 7.8.1 in the windows dev version.
solved issues in occt later than 7.8.1 can not do their work.
In Linux libpack 3.5.5 available to introduce occt 8.0.1 into the work.

luzpaz, 2026-09-12:
>  In Linux libpack 3.5.5 available to introduce occt 8.0.1 into the work.

Isn't libpack Windows exclusive (has nothing to do with Linux) ?

stefano2734, 2026-09-12:
> > In Linux libpack 3.5.5 available to introduce occt 8.0.1 into the work.
> 
> Isn't libpack Windows exclusive (has nothing to do with Linux) ?

Yes, that is here often the problem of windows users and devs.
They use an old occt version and want solved issues of the later versions of occt.
What goes wrong in windows dev packaging in 26.3 with old occt 7.8.1?
7.9.3 and 8.0.1 are available.