FreeCAD/FreeCAD #19333: Light Theme: button text cut off when exiting without save

furgo16, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open two documents with FreeCAD
2. Modify both, without saving them
3. Exit FreeCAD

See the text cut off in the <kbd>Close without Saving</kbd> button:

[image]

With FreeCAD Classic the text is in full:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40038 (Git) Snap 1287
Build type: Release
Branch: main
Hash: 0d96954bcaa7a3de7d74d3804d52bde5292658ac
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-31:
`Apply answer to all` could really be shortened to `Apply to all`

Syres916, 2025-01-31:
Other non-default stylesheets handle it OK, so PR submitted https://github.com/FreeCAD/FreeCAD/pull/19342

PR #19342 (2025-02-02): [Gui] Fix Close without Saving button size
Fixes #19333 

After fix applied:

[image]

[image]

Also fixed two typos in FreeCAD Light which also fixes #19332
closed 2025-02-02 by chennes