FreeCAD/FreeCAD #19337: Freecad stable and 1.1 dev crashes on "polar pattern" with added feature.

venzislav, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I make a hole and chamfer on one of the sides and after this I make "polar pattern" of the hole and add the chamfer feature it works. 
But if add second ring of holes in same manner and if chamfer of second ring of holes interact with chamfer from first ring of holes the app crash.
I found work around - if I reduce the size of the chamfers to smaller size that ther don't interact between the ring it does not crash.
Here is example :

[image]

### Full version info

```shell
it crash on beta and on stable the same way
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-01-31:
Please attach file

venzislav, 2025-01-31:
I will but : File type not allowed: .FCStd  ?

tarman3, 2025-01-31:
Pack to zip

venzislav, 2025-01-31:
Try it to change 2 chamfers that are close to 1mm because now I work around it with 0.9 mm.
When you change on the second ring that intersect it will crash.

[chetkata.zip](https://github.com/user-attachments/files/18619387/chetkata.zip)

luzpaz, 2025-01-31:
@venzislav we need the full [About](https://wiki.freecad.org/About) info of Stable and Development versions you tested on. Thank you.

venzislav, 2025-01-31:
> [@venzislav](https://github.com/venzislav) we need the full [About](https://wiki.freecad.org/About) info of Stable and Development versions you tested on. Thank you.

Here beta :

OS: Ubuntu 24.04.1 LTS (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.53

here stable :

OS: Ubuntu 24.04.1 LTS (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.53

tarman3, 2025-01-31:
To much holes
FreeCAD freeze ))

tarman3, 2025-01-31:
With Draft Array computation only few seconds and file size less

[image]

[chetkata_tarman.zip](https://github.com/user-attachments/files/18620415/chetkata_tarman.zip)

FEA-eng, 2025-09-27:
For me, it freezes indefinitely.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```