FreeCAD/FreeCAD #19338: No shading when accessing the sketch while body visiblity is disabled ( Shading / Lighting issue)

ElektroBOXua, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Whole scene goes in **No shading** mode when accessing sketches if body visiblity is off

Steps to reproduce:
- Create an empty body in Part Design;
- Create sketch;
- Extrude or other operations;
- Disable body visiblity;
- Enter sketch editing mode;
- Exit sketch editing mode;
- Enable body visiblity.

After this body shadings no longer visible, same as in No shading draw style mode.
This bug affects all other objects on the scene!

Changing/playing with render settings didn't do a thing.
Changing draw style modes and object visiblity properties - sometimes works.
Turning flag "**Disable shading in edit mode**" off in general sketcher settings looks like **SOLVED ISSUE** for me.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-01-31:
Would you mind providing a sreencast of the issue ?

ElektroBOXua, 2025-02-01:
[image]

I though it was pretty clear from my description. Whatever.
You see, when i edit/enter/exit sketch of body with visiblity disabled - this happens.
Any scene object gets affected and loses its shades. I already observed this bug on two different devices.

maxwxyz, 2025-02-03:
Could you share an example file?
@kadet1090 also something the 3 point light setup could fix

closed 2025-02-24 by chennes