FreeCAD/FreeCAD #19340: BIM: opening solid visible when wall with window is dragged to IFC project

semhustej, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- create new file
- create Arch_Wall in the file and place Arch_Window in this wall
- create and IFC project inside current file using BIM_project tool
- drag and drop the wall to the IFC project
- the result is an opening solid visible and multiple errors

```
17:04:47  Traceback (most recent call last):
17:04:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40006-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_viewproviders.py", line 337, in <lambda>
17:04:47      QtCore.QTimer.singleShot(100, lambda: self.onDrop(incoming_object))
17:04:47                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
17:04:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40006-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_viewproviders.py", line 343, in onDrop
17:04:47      ifc_tools.aggregate(incoming_object, self.Object)
17:04:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40006-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 1216, in aggregate
17:04:47      aggregate(child, newobj)
17:04:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40006-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 1193, in aggregate
17:04:47      create_relationship(obj, newobj, parent, product, ifcfile, mode)
17:04:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40006-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 1380, in create_relationship
17:04:47      set_attribute(ifcfile, product, "Name", "Opening")
17:04:47                             ^^^^^^^
17:04:47  NameError: name 'product' is not defined

```

See behavior in the video:

https://github.com/user-attachments/assets/9646a623-f789-46db-8e8f-d17b8c81f0ee

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

BIM

### Anything else?

The behavior was discovered by another user in this forum thread: https://forum.freecad.org/viewtopic.php?p=808179#p808179

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-01-31:
It would look like the `product` variable is being used before being assigned here:

https://github.com/FreeCAD/FreeCAD/blob/64a4f6e3669279e5dbe9cb59d8e0123eaeb70833/src/Mod/BIM/nativeifc/ifc_tools.py#L1380

furgo16, 2025-02-01:
@semhustej as a quick test, you could try to change this line on the `ifc_tools.py` file on your installation (around line 1380):

```diff
- set_attribute(ifcfile, product, "Name", "Opening")
+ set_attribute(ifcfile, opening, "Name", "Opening")
```

semhustej, 2025-02-02:
> [@semhustej](https://github.com/semhustej) as a quick test, you could try to change this line on the `ifc_tools.py` file on your installation (around line 1380):
> 
> - set_attribute(ifcfile, product, "Name", "Opening")
> + set_attribute(ifcfile, opening, "Name", "Opening")

I did a test after the change and the result is good:

[image]

It seems this foxed the issue. Thank you.

furgo16, 2025-02-02:
Submitted https://github.com/FreeCAD/FreeCAD/pull/19364

PR #19364 (2025-02-05): BIM: fix unassigned variable reference
Needs to be thoroughly tested.

Fixes: #19340
closed 2025-02-05 by yorikvanhavre