FreeCAD/FreeCAD #19344: 3D connexion spacemouse

xlavcx, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

spacemouse was working fine with previous release. 
just install release 40041 and display object flicks when acting on spacemouse

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/113.268
Installed mods: 
  * fasteners 0.5.33
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

BERT-GER97, 2025-02-01:
I've had the same error with the SpaceMouse under WIN11 since dev 40006. With dev 39980 there is no such problem.
Bert

Kastroz, 2025-02-02:
Moving in 3d space the 3dconexion Spaceball image serious trouble Nvidia GeForce GTX 1050 Ti - windows 11
Also Zoom in and out gets inverted from 3d space to 2d space

Thank you all

closed 2025-02-03 by maxwxyz