FreeCAD/FreeCAD #19345: BIM: subvolume stuck at origin when inserting custom preset window

furgo16, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Uncompress the [test files](https://github.com/user-attachments/files/18628400/Test.window.placement.zip)
2. Copy the `F4-preset-github.FCStd` window preset file to `$FREECAD_USER_DATA/Arch/Windows/Custom`. Adjust the path according to your platform.
3. Open the `TestWindowPlacement.FCStd` file with FreeCAD.
4. Switch to the BIM Workbench
5. Press the Arch Window command button
6. On Tasks panel, presets dropdown, select the `Custom - Windows - F4-preset-github` window preset, as installed on step no. 2
7. Place the window on the wall. It works best when "Snap near" is active.
8. Expected: the window is positioned and there is a hole in the wall for its opening
9. Actual: the window is positioned, but its subvolume remains at the axis origin, away from the window location. The subvolume cannot create the hole in the wall

https://github.com/user-attachments/assets/3a7ee210-77d2-469f-ac72-9681a9af7193

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Snap 1289
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * parts_library
  * Arch
  * Assembly4 0.50.15
```

### Subproject(s) affected?

BIM

### Anything else?

This issue seems to happen only when the window preset's `Subvolume` property is being set with a custom subvolume. If no subvolume is defined, the hole is punched through the wall as expected.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-01:
Once again I am convinced that the coordinates of nested objects should always be linked and broken only by force by the user.
Now the situation is the opposite - nested objects are separated and the user must come up with algorithms to link all the objects together.

This problem is almost not typical for the PD workbench. Because there are no separate coordinates. The base is always linked to the descendant by one position. That is why PD is the most efficient workbench.

paullee0, 2025-02-01:
There seems 2 approaches to resolve this:

1. 1st is quick solution:  Map the SubVolume Sketch to the Window's Base Sketch by ObjectXY, done :)
2. 2nd:  Need to tweak Windows Tool to move Window Base Sketch together with SubVolume's relative Placement (not necessarily SubVolume's Sketch - the SubVolume could be created by hundreds of different ways)

See attached model file.  The 2 Windows to the right hand side were created with 1st approach above.

One inserted with Windows Tool, another just copy and past, and Draft Move.

[Test.window.placement.zip](https://github.com/user-attachments/files/18629500/Test.window.placement.zip)

[image]
[image]
[image]
[image]
[image]

furgo16, 2025-02-01:
Thanks!

> 1st is quick solution: Map the SubVolume Sketch to the Window's Base Sketch by ObjectXY. Done :)

I did try to do this at some point, but it didn't work out. Well, from your tests it seems I did something wrong, so I'll look at your example in detail.

While a great workaround, I think 2. should be the approach, as it would be more intuitive and consistent with the behavior of window instantiation by using File > Merge document... to import the preset and then positioning it.

paullee0, 2025-02-02:
Another related issue is how Draft Move works with SubVolume then.

Does it also move SubVolume when user select the Window only?

furgo16, 2025-02-02:
If the window is inserted from a preset with the window tool, as per the issue description, selecting the window and performing `Draft_Move` moves only the window. The subvolume is still stuck at 0, 0, 0.

If the window is inserted with `Std_MergeProjects`, selecting the window and performing `Draft_Move` moves the window and the subvolume.

paullee0, 2025-02-02:
Thanks, I have a quick thought but not sure why the said difference at the moment.  If there is a video recording the detailed workflow, that would be very helpful.  Or I may have a better look when there is a gap :)

yorikvanhavre, 2025-02-05:
One quick fix we could do here is, in the window object, when the placement changes, detect if we have a subvolume object, and if yes move the subvolme too

furgo16, 2025-02-05:
That'd work for me :-)