FreeCAD/FreeCAD #19346: On Linux Mint Cinnamon, main window is not a window, it replaces and breaks the desktop !

Zener131, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Installed snap FreeCAD 1.0.0 because the PC has an old stable Linux Mint 18.2 and I don't have time now to redo all the tunings for a newer one.

It starts correctly but the main  window is abnormal.
- it has no standard titlebar commands to minimize, maximize, close etc.
- the rest of Linux Mint Cinnamon becomes unavailable: no taskbar etc. Only Alt+Tab allows to switch to any other open program.
- FreeCAD can be closed only using its menu "Quit", in this case the window background has replaced the desktop, only the Cinnamon menu appears to be regenerated (like after a crash of Cinnamon) but it no longer works.
- So after just starting and closing FreeCAD on it, the only getaway is the PC's reset button !

Any solution to make it behave as a normal windowed application living only inside its window ?
Thank you.

### Full version info

```shell
snap FreeCAD 1.0.0
Linux Mint 18.2 / Cinnamon
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-01:
Did you try AppImage
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2025-02-01:
Full [About](https://wiki.freecad.org/About) info please.

luzpaz, 2025-02-01:
Also can you test the Snap edge channel ?  (https://wiki.freecad.org/Ubuntu_Snap#Changing_between_different_Snaps)

Syres916, 2025-02-01:
@luzpaz Linux Mint 18.x went out of support in 2021, we've got to class it as best endeavors

luzpaz, 2025-02-01:
> Linux Mint 18.x went out of support in 2021, we've got to class it as best endeavors

Oh, totally missed that.

Zener131, 2025-02-01:
> Full [About](https://wiki.freecad.org/About) info please.

I could pass it to an editor before the hang:

OS: Ubuntu Core 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

luzpaz, 2025-02-01:
@Zener131 again, please test the appimage[^1]

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Zener131, 2025-02-02:
> [@Zener131](https://github.com/Zener131) again, please test the appimage[1](#user-content-fn-1-9e232a2b9a660b8176edd1135dee4981)
> ## Footnotes
> 
>     1. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds [↩](#user-content-fnref-1-9e232a2b9a660b8176edd1135dee4981)

Done  (I used FreeCAD_weekly-builds-40041-conda-Linux-x86_64-py311.AppImage  on the same PC).

**This is much better**: on exit it does not break Cinnamon (tested with several runs).

But it is still impolite: while running, the window is maximized and impossible to reduce or iconize, and it covers the Cinnamon taskbar (even when not set for autohiding) and the menu. The menu can be called only with key "Super" (=Windows on the keyboard). So it appears in its place (leaving the gap for the taskbar) but the taskbar does not.
And so the other applications open are accessible only with Alt+Tab.

[image]

[image]

**Thank you for your help !** It makes FreeCAD usable for me now on this dedicated PC. 

Then as a Qt developer too, I don't understand how this conflict with the desktop can happen, starting with a standard application core... Nor do I understand why the window has no minimizing/maximizing/closing icons... unless the window is just too wide ? Wrong calculation of position ? 
Else Qt internals through outdated libraries ? This is the tricky point: I cannot upgrade the kernel, else this is the return from hibernating that is broken.

I will wait for this change being part of a next stable version.
So I think the issue becomes "worked around" in the meantime ?

Best regards.

luzpaz, 2025-02-02:
@Zener131 if you double click on the titlebar does it fully minimize the window or does it partially minimize it ?

Zener131, 2025-02-03:
> [@Zener131](https://github.com/Zener131) if you double click on the titlebar does it fully minimize the window or does it partially minimize it ?
> 
> Edit: If you're on Linux Mint v18.0[1](#user-content-fn-1-25bdfdecf98f5dbd369d683e1321e6e3) that's from 2016 and i'm sure there are 3rd party depedency problems that will arise because of that.
> ## Footnotes
> 
>     1. https://en.wikipedia.org/wiki/Linux_Mint#Release_history [↩](#user-content-fnref-1-25bdfdecf98f5dbd369d683e1321e6e3)

Hello.

I tried double-clicking everywhere on the main window, it does nothing.
As I remember, this was a good reason to add menu commands to iconize and restore the window in the menu, it was helpful when the icons managed to go out of the screen.

Yes, one day I will have to rebuild this PC from scratch to have the most recent Linux Mint on it, but it means to redo all the tunings then to reinstall and reparameterize all the applications, this is weeks of hard job. Not the stability I expected from Linux, precisely I had thrown Windows away after XP for that reason.  :-(