FreeCAD/FreeCAD #19347: Deleting linked objects breaks FreeCad

leoheck, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Deleting linked objects breaks FreeCad.
Freecad always crashes on simple tasks like deleting objects. 
This kind of behavior has happened since forever.

Isn't there any way for FreeCAD to check if it can delete an object before trying to do it and breaking it?

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7d9d28845320]
#1  0x7d9d2a0294d8 in QTreeViewPrivate::viewIndex(QModelIndex const&) const from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x48
#2  0x7d9d2a02cea4 in QTreeView::visualRect(QModelIndex const&) const from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0xa4
#3  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5(+0x3bf75c) [0x7d9d29fbf75c]
#4  0x7d9d29fcb686 in QAbstractItemView::viewportEvent(QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x346
#5  0x7d9d28eb68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5+0x7b
#6  0x7d9d29d7039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#7  0x7d9d2bb74c99 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0xc9
#8  0x7d9d28eb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5+0x108
#9  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5(+0x1759e6) [0x7d9d29d759e6]
#10  0x7d9d29d766b4 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x474
#11  0x7d9d29d76e7c in QApplicationPrivate::sendSyntheticEnterLeave(QWidget*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x26c
#12  0x7d9d29daa690 in QWidgetPrivate::hideChildren(bool) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x140
#13  0x7d9d29daa807 in QWidgetPrivate::hide_helper() from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x117
#14  0x7d9d29dae9a8 in QWidgetPrivate::setVisible(bool) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x368
#15  0x7d9d2be12b6f in Gui::TreeWidget::onUpdateStatus() from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x7ff
#16  0x7d9d2be14420 in Gui::TreeWidget::checkTopParent(App::DocumentObject*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0xf0
#17  0x7d9d2c164331 in Gui::SelectionSingleton::checkSelection(char const*, char const*, char const*, Gui::ResolveMode, Gui::SelectionSingleton::_SelObj&, std::__cxx11::list<Gui::SelectionSingleton::_SelObj, std::allocator<Gui::SelectionSingleton::_SelObj> > const*) const from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x4c1
#18  0x7d9d2c16537d in Gui::SelectionSingleton::isSelected(char const*, char const*, char const*, Gui::ResolveMode) const from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x11d
#19  0x7d9d2c165aeb in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x10b
#20  0x7d9d2c16db37 in Gui::SelectionSingleton::slotDeletedObject(App::DocumentObject const&) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x3d7
#21  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so(+0x23f09f) [0x7d9d2aa3f09f]
#22  0x7d9d2ac60740 in App::Application::slotDeletedObject(App::DocumentObject const&) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so+0x10
#23  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so(+0x23f09f) [0x7d9d2aa3f09f]
#24  0x7d9d2aa5b90b in App::Document::removeObject(char const*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so+0xfb
#25  0x7d9d2aad9d67 in App::DocumentPy::removeObject(_object*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so+0x77
#26  0x7d9d2aada004 in App::DocumentPy::staticCallback_removeObject(_object*, _object*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADApp.so+0x24
#27  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7d9d2b206068]
#28  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7d9d2b1e5b5c]
#29  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7d9d2b1f1532]
#30  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7d9d2b2abd1d]
#31  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7d9d2b2ab794]
#32  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7d9d2b2cabf3]
#33  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7d9d2b2c669a]
#34  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7d9d2b2ba9c9]
#35  0x7d9d2a5477cc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADBase.so+0x6c
#36  0x7d9d2bbc4c14 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x94
#37  0x7d9d2bbe4c3d in StdCmdDelete::activated(int) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x79d
#38  0x7d9d2bbce60f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x23f
#39  0x7d9d2bbce9cf in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x12f
#40  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7d9d28eeb6f2]
#41  0x7d9d29d6a462 in QAction::triggered(bool) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0x32
#42  0x7d9d29d6ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0xab
#43  0x7d9d2bc2e0e0 in Gui::ShortcutManager::onTimer() from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x100
#44  0x7d9d2bc2f0cd in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x52d
#45  0x7d9d2bc313d6 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0x96
#46  0x7d9d28eb67ca in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5+0x6a
#47  0x7d9d29d703e8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Widgets.so.5+0xb8
#48  0x7d9d2bb74c99 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libFreeCADGui.so+0xc9
#49  0x7d9d28eb6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5+0x108
#50  0x7d9d29562d63 in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Gui.so.5+0x423
#51  0x7d9d2956368b in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Gui.so.5+0x5b
#52  0x7d9d2950e60a in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Gui.so.5+0x16a
#53  0x7d9d2953129f in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Gui.so.5+0x7f
#54  /tmp/.mount_FreeCAKIHoMB/usr/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0xbf1b) [0x7d9d23484f1b]
#55  /tmp/.mount_FreeCAKIHoMB/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7d9d28eeb6f2]
[... 31 lines cut ...]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.26
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-02-01:
Example file for the ticket ?

leoheck, 2025-02-01:
I cannot share this one, sorry.

luzpaz, 2025-02-01:
...and no way for you to reproduce a scaled down example ?

maxwxyz, 2025-02-03:
Could this be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/13501 ?

leoheck, 2025-09-25:
I could not reproduce this anymore with the latest versions of Fc.

Currently using this one.
```
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

closed 2025-09-25 by leoheck