FreeCAD/FreeCAD #19348: Linux builds fail at link time referencing missing symbols from 3DConnexions SDK

johnsonm, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As discussed in [the forum](https://forum.freecad.org/viewtopic.php?p=807783), builds on Linux have recently (as of  #17000) started to fail with references to missing symbols:

```
/usr/bin/ld: src/Gui/CMakeFiles/FreeCADGui.dir/GuiApplicationNativeEventAware.cpp.o: in function `Gui::GUIApplicationNativeEventAware::GUIApplicationNativeEventAware(int&, char**)':
/home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplicationNativeEventAware.cpp:53:(.text+0xb2): undefined reference to `Gui::GuiNativeEvent::GuiNativeEvent(Gui::GUIApplicationNativeEventAware*)'
/usr/bin/ld: src/Gui/CMakeFiles/FreeCADGui.dir/GuiApplicationNativeEventAware.cpp.o: in function `Gui::GUIApplicationNativeEventAware::initSpaceball(QMainWindow*)':
/home/fyr/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplicationNativeEventAware.cpp:62:(.text+0xfc5): undefined reference to `Gui::GuiNativeEvent::initSpaceball(QMainWindow*)'
```

Applying the change in https://forum.freecad.org/viewtopic.php?p=807783#p807800 enables the build.

I would expect the cmake stage to fail if an SDK update were required, or preferably not try to build support that isn't there, rather than configuration succeeding and the build only failing at the final link stage.

### Full version info

```shell
Failure to build means that there is no "About FreeCAD dialog" but I have reproduced this failure in both Fedora 41 with a Qt6 build and Debian with a Qt5 build.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
@pskowronskiTDx FYI

pskowronskiTDx, 2025-02-04:
@johnsonm The following changes should do the job. Sorry for overlooking this! https://github.com/FreeCAD/FreeCAD/compare/main...pskowronskiTDx:FreeCAD:linux-fix

johnsonm, 2025-02-04:
I'll plan to validate this after work today, thank you!

ovo-Tim, 2025-02-04:
@pskowronskiTDx Also works for me. Thank you.

ovo-Tim, 2025-02-04:
Why uncompilable codes can still pass all the PR checks?

johnsonm, 2025-02-04:
@ovo-Tim configuration combinatorics will make this happen from time to time, as things stand. I would suggest that this bug report isn't the place in which you want to offer to provide a large build farm in which every PR can have every build configuration across all variations of all supported OSes run a complete build end-to-end. 😉

closed 2025-02-04 by chennes