FreeCAD/FreeCAD #19351: Missing borders 3D view

MisterMakerNL, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Got the remark on the stylesheets that we are missing a line on the tabs. Well I added this.
But it looks out of place if the rest of the window has no borders.
See:
[image]
How I would it too look like:
[image]
I was looking at https://github.com/FreeCAD/FreeCAD/blob/64a4f6e3669279e5dbe9cb59d8e0123eaeb70833/src/Gui/Quarter/QuarterWidget.cpp
for a possible fix, but I don't know.

### Full version info

```shell
One of the base stylesheets.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-01:
My initial goto would be `QMdiSubWindow` or `QMdiArea` but I have to say you're giving me too much credit there.

Something like (not tested):

`QMdiSubWindow { border-left: 1px solid #000000; border-right: 1px solid #000000; }
`
You may need to have a little padding but I've never actually messed with this before.

MisterMakerNL, 2025-02-01:
Qmdisubwindow only affects undocked windows.
QMdiArea seems to work but crashes Freecad. Something is also overwriting it. I wonder if Qframe does this.

[image]

maxwxyz, 2025-02-01:
It should not have a border if the view is not full screen (e.g. tiled windows)

MisterMakerNL, 2025-02-01:
@maxwxyz don't you think a small 1px line does complete the look better?

This kinda worked, but also not really. It gives every widget in QmdiArea borders. 

QMdiArea QFrame {
    border-left: 1px solid #ababab;
    border-right: 1px solid #ababab;
    /* border-top: 1px solid #ababab; */
 }
But the issue is that is gives every widget in that area
[image]
Startpage looks like this then:

[image]

maxwxyz, 2025-02-01:
Yes but not with multiple 3D views (View>tiled windows)

Syres916, 2025-02-01:
@MisterMakerNL when the important rugby match has finished I'll take a deeper look but [this](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/MainWindow.cpp#L1119) is where I'll be looking.

MisterMakerNL, 2025-02-01:
Hope your team wins!!
I got this far:
The first code I set it too on, and the second and third I try to remove it from the parts where I don't want it.
Which is a very dirty way of doing it. But I cannot seem to find what widget I have to point toodirectly!

QMdiArea QFrame {
    border-left: 1px solid #ababab;
    border-right: 1px solid #ababab;
 }
 QMdiArea QLabel {
    border-left: 0px solid #d0c711;
    border-right: 0px solid #5900ff;
 }
  QMdiArea QListView {
    border-left: 0px solid #e47616;
    border-right: 0px solid #432182;
 }

Syres916, 2025-02-01:
England lost but that's another story. Yep, this one's not easy to do 'properly', what we're trying to do is:

```
        mw = Gui.getMainWindow()
        for w in mw.findChild(QtWidgets.QMdiArea).findChildren(QtWidgets.QWidget):
            if w.inherits("SIM::Coin3D::Quarter::QuarterWidget"):
```

I'm certainly struggling, maybe a fresh day might be inspiring!!

MisterMakerNL, 2025-02-01:
I am also getting nowhere tried using # to point to Quarterwidget without success. 
It's frustrating since I am touching it but cannot grab it.

Syres916, 2025-02-01:
Try using [this](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Only_Border_Coin_Based_Window_dev) as a base

MisterMakerNL, 2025-02-01:
> Try using [this](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Only_Border_Coin_Based_Window_dev) as a base

Tried this, looks good but it didn't work!

kadet1090, 2025-02-01:
QFrame is very basic widget that is used as base for many other, we should be very careful styling it. If possible we should have less generic selector like `QMdiArea > QFrame`.

MisterMakerNL, 2025-02-02:
Draft PR that is draft because I hope we can fix this too!
https://github.com/FreeCAD/FreeCAD/pull/19367

MisterMakerNL, 2025-02-13:
I can do this with the following ugly code:
```
Gui--MDIView QFrame{
  border-radius: 1px;
  border-left: 1px solid #d8d8d8;
  border-right: 1px solid #d8d8d8;
  border-top: 1px solid #d8d8d8;
  border-bottom:0px solid transparent;
}

Gui--MDIView QLabel{
border: 0px;
background-color: transparent;
}
Gui--MDIView QScrollArea{
  border: 1px solid #d8d8d8;
background-color: transparent;
}
```

closed 2026-10-19 by MisterMakerNL