FreeCAD/FreeCAD #19352: Sketcher tool menus are often invisible on Qt6 build

johnsonm, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On a Qt6 build of 1.1dev on Fedora 41 (Qt5 does not build there due to not having older pyside), the sketcher tool menus are usually but not always invisible. The only way to tell which sub-tool is selected is the hovertext:

[image]

[image]

[image]

This is inconsistent. For example, here are two immediately sequential attempts to choose a fillet tool mode from the same tool:

[image]

[image]

No console errors are produced when these menus fail to appear.

### Full version info

```shell
OS: Fedora Linux 41 (Toolbx Container Image) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git)
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.13.1, Qt 6.8.1, Coin 4.0.2, Vtk 9.2.6, OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 96/225.585
Installed mods: 
  * Assembly4 0.50.15
  * Curves 0.6.47
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.29
  * MeshRemodel 1.8919.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * parts_library
  * Defeaturing 1.2.2
  * A2plus 0.4.67
  * CurvedShapes 1.0.10
  * Manipulator 1.5.7
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.63.0
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-01:
@MisterMakerNL you have a qt6 build?

johnsonm, 2025-02-01:
Aha! It appears that this is specific to wayland. It reproduces when I `export QT_QPA_PLATFORM=wayland` but works correctly if I `export QT_QPA_PLATFORM=xcb`

So no great surprise that it's not reproducing on Windows.

maxwxyz, 2025-02-03:
@kadet1090 any idea?

karliss, 2025-03-12:
Did some testing. Seems like it might be some kind of redraw related race condition. 

My initial observation was that on  main monitor (2560x1440) dropdown fails to open ~50% of time, but on my right monitor (4K) it fails 80% of time and on my leftmost monitor (1920x1080) it fails <10%.  When FreeCad window isn't maximized dropdown mostly succeeds. This gave me idea that it might be related to window size, since the order of failure rate also matches my screen resolutions.

Having window resized so that it covers 90% of screen but isn't maximized produced similar error rate as having window maximized. Reducing window size to <1/4 the screen result in high success rate even on my 4K screen. This more or less confirms correlation between window size and failure rate.

Having some kind of race condition would also match with the observed the inconsistency -> the menu sometimes opening successfully and sometimes failing.

Additional observations: 
* Problem isn't exclusive to Sketcher. I can repeat it with more or less any toolbutton which is using `setPopupMode(QToolButton::MenuButtonPopup)`. For example PD buttons and even common buttons like those in "View" toolbar.
* those toolbar buttons with dropdown are using more or less standard QToolButton . Other Qt software on my system also using  QToolButton popup isn't affected by the bug. There are a few other FreeCad Wayland bugs which don't seem to affect other Qt programs using the same standard widgets. For example the one with sticky toolbar dragging. Wouldn't be surprised of multiple of them were caused by the same problem. But without additional investigation hard to be sure if it's something specific to FreeCad setup or other Qt programs simply don't hit the timing requirements for the bug.

karliss, 2025-03-15:
Related #17982

johnsonm, 2025-03-16:
For clarity: I don't have the resize problems described in #17982 in wayland and I do continue to see this. (I once had a resize problem in a previous build but was unable to reproduce it to report.) I am running GNOME. I currently have built version ba2a377

luzpaz, 2025-05-17:
@johnsonm @karliss can you still reproduce on latest 1.1dev ?

karliss, 2025-05-17:
Yes, just compiled 61e349ef54c4f15316f7851cf0dbbae4b82fd3a9 and it still happens.

johnsonm, 2025-09-25:
I confirm that  hash 929bd5090be013d41dc1216cd9bc4d9b0283c41d is still affected.