FreeCAD/FreeCAD #19355: accessibility problem

IDADV, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello I have a problem I am visually impaired and I would like to use freeCAD but unfortunately it is not suitable for me so how could I make the buttons and text of the software suitable for me

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu: GNOME/ubuntu/xcb)

Architecture:

x86_64

Version: 1.0.0.38641 +468 (Git) Snap 1248

Build type: Release

Branch: tag: 1.0.0

Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-01:
Could you tell us what need to be changed? Is it size, contrast, placement,...?

IDADV, 2025-02-05:
Good evening, I will try to make a design for you on figma to show you what we could modify, thank you very much and have a good evening.

IDADV, 2025-02-15:
hello I am following up on our discussion that we had for the graphical interface of freecad
so I warn you in advance these are only photos that I took with my phone that's why there is a little light so I will show you the first page when you arrive on the software you see at the top the buttons that will allow you to create the types of 3D files but I did not write on it I will let you design it in your own way here is an image

[image]

then on the second photo it is still part of the first page when we go down further we see all the files that we created with the black frame which will represent the modeling that we will have done therefore an image and the title at the bottom

[image]

then on the third photo we see the part where we can do modeling or in particular on the left side a small bar where there are three buttons which are represented with shapes the first is used to open a menu on the side we can choose to open either the command prompt or other things on the far right then the third button is used to open a menu also where we can choose the tools we want to use like a sketch... and finally the third button which will also open a menu where we can save or open the file to do lots of other things here are the images

[image]

[image]

I think the ideal font size for the whole software would be 24 and if you agree to redevelop the graphical interface I would like to test it after development to tell you if it is all good before publishing it if you do not mind thank you very much and have a good day

IDADV, 2025-02-15:
and the most important thing in the software for it to be adapted for visual impairment is not only the design, it must also be well organized

IDADV, 2025-02-22:
Cc @maxwxyz

IDADV, 2025-02-28:
hello @maxwxyz  is it all good for the design?

maxwxyz, 2025-04-10:
I think the idea and pain points are understood but I'm not sure if this can be implemented quickly.
@FreeCAD/design-working-group FYI