FreeCAD/FreeCAD #19357: [v1.1dev] Spreadsheets: basic inputs broken

eblanshey, 2025-02-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In 1.1dev released 2025-01-30 basic spreadsheet input is broken.

Scenario 1:

1. Create spreadsheet
2. Start typing `=1`
3. The cell exits typing mode by itself, and saves "1"

The above happens when typing =0, =1, and =2, but not =3 and on.

Scenario 2:

1. Create spreadsheet
2. Paste the text `=100mm` from another program, e.g. vscode
3. Press enter
4. Double click the cell to edit it.
5. Put cursor after the second 0
6. Press backspace
7. The cell exits typing mode by itself and saves `10mm`

Note that scenario 1 does not always happen for me, and scenario 2 does. But after doing scenario 2, I can usually start triggering scenario 1.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/140.99
Installed mods: 
  * WebTools 1.0.0
  * Help 1.0.3
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-02-02:
It is reproducible also on _Windows_. I failed to reproduce it with the stable version.

xtemp09, 2025-02-02:
This issue is connected to #16987 or even the same.

luzpaz, 2025-02-02:
Closing as  dupe of #16987

closed 2025-02-02 by luzpaz

eblanshey, 2025-02-23:
@xtemp09 just want to say thank you for fixing this issue!