FreeCAD/FreeCAD #19359: Arch Transparent Window

ukharwa, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Model window is transparent. Unable to see model however you can still interact with it.

### Full version info

```shell
OS: Arch Linux (Hyprland/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-02:
Wayland is the important factor here IMHO.

quinton-c, 2025-02-02:
I had this pop up earlier today too, whereas it was working fine a few days ago. Updated my pc, and now freecad won't open, which might be a separate issue.

closed 2025-02-03 by maxwxyz