FreeCAD/FreeCAD #19360: Cannot access folders with special caraters such as '#'

KiudLyrl, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

if a folder contain a special character such as a '#' freecad will say that he cannot save the file because it cannot find the parent folder.

issue appears on windows 10

### Full version info

```shell
1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
Please share full about info and steps how to reproduce

luzpaz, 2025-11-25:
> if a folder contain a special character such as a '#' freecad will say that he cannot save the file because it cannot find the parent folder.

Can someone on windows 10/11 reproduce ?