FreeCAD/FreeCAD #19361: BIM: cannot add

furgo16, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When trying to place a window using links, the original window needs to be in the same document as the links to that window. Otherwise adding the window to a wall does not work.

That is, if you have a file called "My window preset.FCStd" and then insert it as an external link to a "My house.FCStd" file, then it cannot be added to a wall via the Arch_Add command.

This is because the inserted link needs to set its "Hosts" property to be the wall. But the "Hosts" property is one that all links share with the original linked object. Since the original linked object is in an external file, it cannot see the wall within its file scope. 

[LinkWindowTest.zip](https://github.com/user-attachments/files/18631367/LinkWindowTest.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Snap 1289
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * parts_library
  * Arch
  * Assembly4 0.50.15
```

### Subproject(s) affected?

BIM

### Anything else?

There might be some workarounds:
- Setting the `LinkCopyOnChange` property on the Link windows
- Adding a new `Hosts` property on the Link windows

This issue is reportedly not reproduceable if using the `ArchSketch` add-on.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-02-02:
There should be an option to set the "Link Copy On Change" option to "True" by default.
https://github.com/FreeCAD/FreeCAD/issues/14930

This would partially solve the problem.

paullee0, 2025-02-02:
> This is because the inserted link needs to set its "Hosts" property to be the wall. But the "Hosts" property is one that all links share with the original linked object. Since the original linked object is in an external file, it cannot see the wall within its file scope.

See further explanation of the problem here :) - https://forum.freecad.org/viewtopic.php?p=808569#p808569

- Not only about the Hosts in external document only point to that document but not the linked document, the underlying problem with stock Link instance of a Window is - - - All Links instances of a Window SHARE THE SAME SET OF PROPERTIES INCLUDING HOSTS (except Placement of cos and those shown in black colour, those 'belongs' to the Links instances) OF THE LINKED OBJECT (i.e. the Window), so no matters how many Links instance of Window user currently have, they can only point to and share 1 single Host object (Wall e.g.), AND,  at the same time, Links in linked document suffer from the 'scope' of the original Linked Window, which is in the external Document.

The feature to support 1/ each Links instance of Windows has their own individual setting of Hosts, and 2/ it see the 'scope' of the Links in its document (rather than the Linked Window in the external document) was already in the external SketchArch add-on  ( I always thought it had been in the stock FreeCAD Arch/BiM).

I'll see how best to migrate that :)

furgo16, 2025-02-02:
What's the best solution here then, setting `Link Copy On Change=True` (or having a mechanism to remember the last setting as per https://github.com/FreeCAD/FreeCAD/issues/14930#issuecomment-2197505188), or adding a `Hosts` property to each link?

paullee0, 2025-02-02:
Hosts property is inherent to Windows object, it would be intuitive and enough for user to use it if it is particularly 'enabled' for Links (of Windows).

Link Copy on Change etc. is in fact was developed on one hand for producing variants, not for memory and model size efficiency, it is an overkill and more convoluted in this case IMHO.  See wiki or excerpt below.

Wiki - https://wiki.freecad.org/Std_LinkMake :-

1. Enabled - "... After the deep copy is performed, there won't be any linkage between the original and the copied object. Therefore, changes in the original object won't be reflected in the copies."
2. Tracking - "Same as Enabled, but additionally the copy will be automatically refreshed if the original source object changes."

Incidentally, I am developing from another approach variants with Variant PropertySet, no need of DeepCopy etc., this is another story.

kaiwas, 2025-02-02:
> not for memory and model size

It would be possible to change it. But if you don't use this parameter, then to create a "unique" object you will need to copy and paste. And this also increases memory consumption and file size.
I think it would be better if the user had a way to use this setting without turning it on every time they create a new link.

paullee0, 2025-02-03:
PR submitted - https://github.com/FreeCAD/FreeCAD/pull/19378

closed 2025-02-14 by yorikvanhavre