FreeCAD/FreeCAD #19363: [TNP] [Assembly] Toponaming problem still present in assembly ?

mgth, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

here the two part are joined by 3 Distance joins
[image]

but if I drill a hole on the side of one : 

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: ProDark.qss/ProDark/Fusion
Logical/physical DPI: 96/93.338
Installed mods: 
  * Assembly3 0.12.2
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * FreeCAD-themes 2024.7.24
  * freecad_streamdeck_addon 0.1.7
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * workfeature
```

### Subproject(s) affected?

None

### Anything else?

here is the file used : 

[Assembly Toponaming Bug.zip](https://github.com/user-attachments/files/18631746/Assembly.Toponaming.Bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ftdt10, 2025-02-03:
I tested the file, but no such problem.

OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40043 (Git)
Build type: Release
Branch: main
Hash: 34d9ac00506ace1b930a4efa075b97bee810f1c7
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/164.765
Installed mods:

mgth, 2025-02-03:
> I tested the file, but no such problem.

Your right, if I try the file again it works.
In fact the problem is there only if done from scratch (I tried it again), even if I save before doing the pocket. 
But if I close and reopen the file before doing the pocket the problem is gone.

More if I save after the pocket when the problem is there, if I reopen the file the problem is still there, even with assembly recalculation.

maxwxyz, 2025-02-03:
Can you record a macro of how you create the file and share it? Does it work in the stable 1.0 version?

mgth, 2025-02-04:
Yes it does the same in stable 1.0 version.

I'm trying to make a macro, but unfortunately the resulting macro is very buggy and wont play back.
I tried debugging it, but until now, I could not get Assembly_CreateJointDistance to work.

Footz355, 2025-02-05:
Greetings. Just stumbled on this issue and it seems like I have similar issue in Assembly after applying Hole with countersink attribute on one face.
Here is an assembly of a hinge I've made:
[image]

After performing "hole" with countersink the assembly looks as follows:
[image]

To reproduce:
See file attached with hinge assembly, and 2 parts of hinge in bodies, Hinge 2 has circles ready to make Hole operation. Perform hole operation on Hinge 2 with countersink attribute. Assembly gets skewed.

See file attached:

[hinge.zip](https://github.com/user-attachments/files/18669007/hinge.zip)

FreeCad info:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

Footz355, 2025-02-07:
> > Greetings. Just stumbled on this issue and it seems like I have similar issue in Assembly after applying Hole with countersink attribute on one face. Here is an assembly of a hinge I've made
> 
> As for me if I try your file I don't get the problem, but if I delete the assembly an recreate it, then it's there.

Just to be sure I have redownloaded my own file, and tried to modify again and still the issue persist on my side:

[image]

I have noticed that sometimes if the assembly gets broken, I do Ctrl+Z, or Ctrl+Z and Ctrl+Y, and you can see the assembly origins move around
before:

[image]

after Ctrl-Z:
[image]

drwho495, 2025-02-07:
Are all of the parts designed in the latest daily build?

Footz355, 2025-02-07:
> Are all of the parts designed in the latest daily build?

Nope, I am using 1.0 stable, had too many bugs and crashes in dev.

mgth, 2025-02-07:
> > Are all of the parts designed in the latest daily build?
> 
> Nope, I am using 1.0 stable, had too many bugs and crashes in dev.

Just tried again, and it breaks in 1.0 but not in dev. But the problem is still there if I rebuild the assembly.

drwho495, 2025-02-07:
> > Are all of the parts designed in the latest daily build?
> 
> Nope, I am using 1.0 stable, had too many bugs and crashes in dev. 

So to my knowledge, if you don't want to deal with tnp in assembly, then you need to design your parts in a newer version of 1.1dev then assemble it. You cannot use older 1.1dev build or 1.0.0 designed parts in assembly if you don't want to deal with TNP

Footz355, 2025-02-08:
Well yeah, the thing is, when I experienced other bugs in dev version, people were telling me not to use dev version, and stick with the stable lol. I'll try dev again, thanks.

eblanshey, 2025-02-08:
> So to my knowledge, if you don't want to deal with tnp in assembly, then you need to design your parts in a newer version of 1.1dev then assemble it. You cannot use older 1.1dev build or 1.0.0 designed parts in assembly if you don't want to deal with TNP

Do you know if there is going to be a way to "migrate" files to the new assembly format? Redoing whole projects isn't a realistic solution.

furgo16, 2025-02-08:
I do not think the Assembly module in either FreeCAD 1.0.0 or 1.1.0dev is claiming to be TNP-tolerant. If it is, it'd be worth pointing at the relevant PR.

drwho495, 2025-02-08:
> Do you know if there is going to be a way to "migrate" files to the new assembly format? Redoing whole projects isn't a realistic solution.

There might be a way to completely recompute a part, im not sure though. But this is a bug that needs to be fixed.

mgth, 2025-02-15:
For more information I noticed that when creating the hole feature :
If I do it from scratch I get, in the report view :
```
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:1;FAC;:H59e:4,F;:H59f,F;:H595:8,F.Face5 -> Body001.Clone.;#a:1;:H59f,F;:H5b3,F;:H595:8,F.Face3
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:1;FAC;:H59e:4,F;:H59f,F;:H595:8,F.Face5 -> Body001.Clone.;#a:1;:H59f,F;:H5b3,F;:H595:8,F.Face3
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint001.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:3;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F.Face3 -> Body001.Clone.;#c:3;:H59f,F;:H5b3,F;:H595:8,F.Face7
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint001.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:3;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F.Face3 -> Body001.Clone.;#c:3;:H59f,F;:H5b3,F;:H595:8,F.Face7
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint002.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:4;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F.Face4 -> Body001.Clone.;#c:4;:H59f,F;:H5b3,F;:H595:8,F.Face2
09:37:21  <PropertyLinks> PropertyLinks.cpp(473): Unnamed#Joint002.Reference1 auto change element reference Unnamed#Clone Body001.Clone.;#3:4;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F.Face4 -> Body001.Clone.;#c:4;:H59f,F;:H5b3,F;:H595:8,F.Face2

```

but If I save and reopen before creating it, I get many more auto changes, and it works :
 09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint.Reference2 auto change element reference Nouveau#Body Body.;#3:3;:G;XTR;:H59e:7,F;:H59f,F.Face3 -> Body.;#c:3;:H59f,F;:H5b0,F.Face7
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint.Reference2 auto change element reference Nouveau#Body Body.;#3:3;:G;XTR;:H59e:7,F;:H59f,F.Face3 -> Body.;#c:3;:H59f,F;:H5b0,F.Face7
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint001.Reference2 auto change element reference Nouveau#Body Body.;#3:1;FAC;:H59e:4,F;:H59f,F.Face5 -> Body.;#a:1;:H59f,F;:H5b0,F.Face3
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint001.Reference2 auto change element reference Nouveau#Body Body.;#3:1;FAC;:H59e:4,F;:H59f,F.Face5 -> Body.;#a:1;:H59f,F;:H5b0,F.Face3
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint002.Reference2 auto change element reference Nouveau#Body Body.;#3:4;:G;XTR;:H59e:7,F;:H59f,F.Face4 -> Body.;#c:4;:H59f,F;:H5b0,F.Face2
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint002.Reference2 auto change element reference Nouveau#Body Body.;#3:4;:G;XTR;:H59e:7,F;:H59f,F.Face4 -> Body.;#c:4;:H59f,F;:H5b0,F.Face2
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:1;FAC;:H59e:4,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face5 -> Body001.;#a:1;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face3
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:1;FAC;:H59e:4,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face5 -> Body001.;#a:1;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face3
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint001.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:3;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face3 -> Body001.;#c:3;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face7
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint001.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:3;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face3 -> Body001.;#c:3;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face7
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint002.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:4;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face4 -> Body001.;#c:4;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face2
09:38:29  <PropertyLinks> PropertyLinks.cpp(473): Nouveau#Joint002.Reference1 auto change element reference Nouveau#Body001 Body001.;#3:4;:G;XTR;:H59e:7,F;:H59f,F;:H595:8,F;:H5a9:12,F.Face4 -> Body001.;#c:4;:H59f,F;:H5b0,F;:H595:8,F;:H5a9:12,F.Face2

Norbert-Rotorwerk, 2025-03-11:
Here is a model showing the toponaming problem in the assembly workbench. 
Steps to reproduce: 
1. Open the file Platte_Schrauben_asm.FCStd
2. Open the file Platte.FCStd
3. add a fillet on the 4 edges of the Platte, e.g. 5 mm
4. go to the file Platte_Schrauben_asm.FCStd
5. activate the assembly
6. right click on the assembly
7. click on recompute object

You can reverse this by removing the fillets of the plate 'Platte', than recompute the assembly again. And it is repeatable. 

[image]

[image]

[Platte_Schrauben_asm.zip](https://github.com/user-attachments/files/19181079/Platte_Schrauben_asm.zip)

Norbert-Rotorwerk, 2025-03-11:
Version 1.0.0 was used.

OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Render 2023.2.2
  * Help 1.0.3
  * FEM_FrontISTR 0.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.37
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.26
  * BillOfMaterials 1.0.2
  * Design456 0.0.1
  * OpenTheme 2024.9.1

drwho495, 2025-03-11:
Ok, so I believe there was a pr that pretty much fixed the assembly issues in newer versions of 1.1dev. The parts have to be designed and assembled in 1.1dev for it to work.

drwho495, 2025-03-11:
This issue is improved in 1.1dev, it just loses reference every single time you make a change.

drwho495, 2025-03-11:
Also it looks like there is a TNP issue that is related to https://github.com/FreeCAD/FreeCAD/issues/14595

Norbert-Rotorwerk, 2025-03-12:
> Ok, so I believe there was a pr that pretty much fixed the assembly issues in newer versions of 1.1dev. The parts have to be designed and assembled in 1.1dev for it to work.

I rebuild the model in 1.1.dev and it happened again. 

[Platte_Buchse_asm_1.1.dev.zip](https://github.com/user-attachments/files/19205717/Platte_Buchse_asm_1.1.dev.zip)

[image]

[image]

[Platte_Buchse_asm_1.1.dev.zip](https://github.com/user-attachments/files/19205790/Platte_Buchse_asm_1.1.dev.zip)

 OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/94.0197
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Render 2023.2.2
  * Help 1.0.3
  * FEM_FrontISTR 0.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.37
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.26
  * BillOfMaterials 1.0.2
  * Design456 0.0.1
  * OpenTheme 2024.9.1

drwho495, 2025-03-16:
> I rebuild the model in 1.1.dev and it happened again.
>
> [Platte_Buchse_asm_1.1.dev.zip](https://github.com/user-attachments/files/19205717/Platte_Buchse_asm_1.1.dev.zip)
>
> [image]
>
> [image] [...]

This is bizarre, because when I do the same test, Freecad acts like it does in every other workbench and usually references go missing. This is a hard issue to fix because it's hard to repeat. Just to check, everything was redesigned in 1.1?

drwho495, 2025-03-18:
> > Ok, so I believe there was a pr that pretty much fixed the assembly issues in newer versions of 1.1dev. The parts have to be designed and assembled in 1.1dev for it to work.
> 
> I rebuild the model in 1.1.dev and it happened again.
> 
> [Platte_Buchse_asm_1.1.dev.zip](https://github.com/user-attachments/files/19205717/Platte_Buchse_asm_1.1.dev.zip)
> 
> [image]
> 
> [image]
> 
> [Platte_Buchse_asm_1.1.dev.zip](https://github.com/user-attachments/files/19205790/Platte_Buchse_asm_1.1.dev.zip)
> 
> OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb) Architecture: x86_64 Version: 1.1.0dev.40504 (Git) Conda AppImage Build type: Release Branch: main Hash: [855ad6c](https://github.com/FreeCAD/FreeCAD/commit/855ad6c70904f911b2ce7114dd1a5bc24da98bd1) Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2 Locale: German/Germany (de_DE) Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default Logical/physical DPI: 96/94.0197 Installed mods:
> 
> * BOLTSFC 2022.11.5
> * parts_library.bak
> * parts_library
> * symbols_library
> * Defeaturing 1.2.0
> * Assembly4 0.12.6
> * Manipulator 1.5.0
> * MeshRemodel 1.8919.0
> * Render 2023.2.2
> * Help 1.0.3
> * FEM_FrontISTR 0.2.0
> * QuickMeasure 2022.10.28
> * fasteners 0.5.37
> * 3DfindIT 1.2.0
> * sheetmetal 0.4.26
> * BillOfMaterials 1.0.2
> * Design456 0.0.1
> * OpenTheme 2024.9.1

Tested on my machine and works as intended.

drwho495, 2025-03-18:
> I just modelled it from scratch on the last 1.1.0dev Revision 40655. I get the same issue as in the models before. 

That's bizzare. Could you make a video of you making the change that causes the tnp?

Norbert-Rotorwerk, 2025-04-22:
Please, find a video of making the issue. 

I used 

OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0197/1
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Help 1.0.3
  * FEM_FrontISTR 0.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.38
  * 3DfindIT 1.2.0
  * sheetmetal 0.7.22
  * BillOfMaterials 1.0.2
  * Design456 0.0.1
  * OpenTheme 2024.9.1
  * Render.backup1744616697.684052 2023.2.2 (Disabled)
  * Render

I make the issue with a fillet and a chamfer. Removing the fillet will go to the original state. With the chamfer the original state will not come back. 

[toponaming_issue_assembly_plate_bushing.mp4.zip](https://github.com/user-attachments/files/19853077/toponaming_issue_assembly_plate_bushing.mp4.zip)

drwho495, 2025-07-19:
to anyone still encountering this issue, i would suggest doing a few things:
1. your assembly files must ALWAYS be loaded when making any change of any form to a model, this is likely a fixable issue.
2. disable the string hasher on all files (including design and assembly files), this causes issues (to do this, select a document then in the property editor disable string hasher, its enabled by default in every file)

i will be looking into fixing both of these issues along with a ton of edgecases in the future (probably before 1.1)

drwho495, 2025-10-05:
can this be retested on the latest version of 1.1dev

closed 2025-10-17 by maxwxyz