FreeCAD/FreeCAD #19366: SpaceBall Button Dialog "Reset" function breaks the SpaceBall button configuration

glabifrons, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I was testing #10073 as requested by the bot when I encountered #17812, but also the below problem.

I pressed the "Reset" button under the list of buttons in the SpaceBall Buttons dialog, and it cleared the list, replacing it with 31 buttons with added labels of "1", "2", etc.
I have a SpaceBall 5000 USB (which has 12 buttons and one internal/hidden switch), so autodetection of the model and number of buttons is now broken, at least for the "Reset" button.

In an attempt to correct this, I tried to select my model.
The pulldown menu (labeled Default) lists only newer devices as if support for SpaceBalls has been removed (there is no entry for SpaceBall 5000, 4000, 2003, etc.).

You can see both in the below screenshot.

[image]

If I manually edit out the entire button definition section of my user.cfg and relaunch FreeCAD, then go into the SpaceBall Buttons dialog and press one of the SpaceBall's buttons, it correctly determines the number of buttons my model has and populates the list (without the odd additional labels mentioned above).

I was able to recover my user.cfg from a snapshot, but the "Reset" function and autodetection of SpaceBall models appears to be broken.

### Expected behavior:
Pressing a button with an empty button list populates the list properly, so the "Reset" button on the SpaceBall Buttons dialog should **clear** the button list, not replace it with a list of buttons that the device does not have. 

The "Default" pulldown should have a "Model" label and should include SpaceBall models that the dialog was named after, such as the 2003, 4000, 5000, and any other supported models.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (MATE/mate/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.31
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Curves 0.6.51
  * freecad.gears 1.3.0

Also tested:

OS: Ubuntu 22.04.5 LTS (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 93/163.331
Installed mods: 
  * fasteners 0.5.31
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Curves 0.6.51
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

I can retest on other models if needed.
I also have a SpaceBall 4000, a SpaceBall 2003, and a SpacePilot.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FEA-eng, 2025-09-27:
@glabifrons I don't have a SpaceBall manipulator. Is the issue still relevant?

glabifrons, 2025-09-27:
Thank you for asking.
I just checked with the latest weekly build (FreeCAD_weekly-2025.09.24-Linux-x86_64-py311.AppImage) and the problem persists.

luzpaz, 2025-09-27:
@glabifrons can you add your full [about](https://wiki.freecad.org/About) info for the record ?

glabifrons, 2025-09-28:
OS: Ubuntu 24.04.3 LTS (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 93/163.331/1
Installed mods: 
  * fasteners 0.5.37
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Curves 0.6.58
  * freecad.gears 1.3.0