FreeCAD/FreeCAD #19368: CAM Simulator (new) Artefakte with komplex Body

PatrikRindlisbacher, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

I've already done a lot of testing including weekly builds and graphics card updates.

Windows 11 with NVidia Quadro P2000
NVIDIA RTX Driver Release 570 R570 U1 (572.16) | Windows 11

I've also had problems in the simulation with smaller parts

[CAM_2025_02_V01.zip](https://github.com/user-attachments/files/18632627/CAM_2025_02_V01.zip)

### Full version info

```shell
1.0.0 and 1.1.0.dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-02-02:
Full [About](https://wiki.freecad.org) info for each version you tested on, please.

JohnOCFII, 2025-02-02:
I see similar display issues on macOS.  I also noted some warnings displayed in the Report Window. 

```
11:21:54  v1.1, Model-2025_02__V01_220517, added 'ForceCompound' property
11:21:54  New property added to "Profile": ['NumPasses', 'Stepover']. Check its default value.
11:21:54  v1.1, Stock-2025_02__V01_220517, added 'ForceCompound' property
```

I also note the old simulator has trouble with this model as well.  It displays errors when attempting to simulate:

```
11:23:40  Traceback (most recent call last):
11:23:40    File "/Applications/Other Apps/FreeCAD-40041.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Simulator.py", line 379, in PerformCut
11:23:40      self.PerformCutVoxel()
11:23:40    File "/Applications/Other Apps/FreeCAD-40041.app/Contents/Resources/Mod/CAM/Path/Main/Gui/Simulator.py", line 331, in PerformCutVoxel
11:23:40      dz = (cmd.z - self.curpos.Base.z) / n
11:23:40            ~~~~~~^~~~~~~~~~~~~~~~~~~~
11:23:40  TypeError: unsupported operand type(s) for -: 'NoneType' and 'float'
```

```
OS: macOS 15.3
Architecture: arm64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 72/127.5
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
```

[image]

[image]

luzpaz, 2025-02-02:
@shaise have you seen this before ?

luzpaz, 2025-05-31:
@tarman3 @shaise any chance you can take a look ?

tarman3, 2025-05-31:
no chance for me  )

shaise, 2025-05-31:
Hi Everyone,

This is most likely happening because of the way I implemented the code. In order for the OpenGL boolean cut "trick" to work flawlessly, I need to order all cut operations in all possible permutations.
this means that say, may G code have 3 operations: A B and C, I need to feed the algorithm with all the possible orders: ABC, ACB, BAC, BCA, CAB and CBA. When a Gcode file can easily contain thousands of operation, this is clearly non practical, hence I resort to feeding the algorithm with only two options. forward and back G1,G2,...Gn and Gn,...G2,G1
This works in most cases, but will probably fail in some, specially the complex ones.
There are some mathematical ways to decrease the amount of permutations needed, and still get flawless result. But this is complex math and I'm yet to find something easy enough to code.

shai

shaise, 2025-07-08:
As mentioned in the forum, the algorithm is
based on papers by Nigel Stewart - http://www.nigels.com/research/
Some of the papers do discuss the performance issue. 
Technically of cut operations can be sorted front to back relative to the camera, it can be done with a single pass. But the sorting is not a simple task.

jffmichi, 2026-10-31:
.