FreeCAD/FreeCAD #19369: Symmetry constraint not working

budiz1, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi, i think i found a problem. I am using a stable version of FreeCad 1.0.0.39109 on MacOS Sequioa and when trying to create a symmetry between two dots the sketch would always turn red showing an error (redundant constrains).
But this is solvable when redundant DoF(s) are deleted, the sketch would position itself creating a symmetry.

I heard that on Windows this feature work fine.

Best regards

### Full version info

```shell
OS: macOS 15.3
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

budiz1, 2025-02-02:
If anyone is able to replicate this bug please report back.

frankfont, 2025-02-02:
Works fine on my MacOS - perhaps you are selecting the wrong thing?

I can get that error IF I do the following:

Create a rectangle, select the >< symmetry icon in the toolbar, click top left dot of my rectangle, click bottom right of my rectangle -- then ( the problem ) click vertical axis. This creates the error messages you describe.

However, if instead of on my third click selecting the axis, I select the origin point -- it works fine! The rectangle gets centered around the origin.

budiz1, 2025-02-03:
[image]
[image]
[image]

Here are the steps that i usually do for the symmetry.

budiz1, 2025-02-03:
Here are the steps that i usually do for the symmetry.

[image]

[image]

[image]

maxwxyz, 2025-02-03:
I also am not able to reproduce. Can you share a demo file?
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.32
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

budiz1, 2025-02-03:
[Symmetry.FCStd.zip](https://github.com/user-attachments/files/18643120/Symmetry.FCStd.zip)

When i grab the lines they snap on the axis making it symmetry. But it stays red (with redundant constraints).

When i shared this file on Windows computer symmetry fixes itself and the lines become white.

FEA-eng, 2025-02-04:
Do you have "Auto remove redundants" enabled in the Sketcher preferences?

budiz1, 2025-02-04:
Hey, that fixed the problem.

closed 2025-02-04 by budiz1