FreeCAD/FreeCAD #19370: BIM/TechDraw - include Doors and Windows when selecting Site/Building/Level as a basis for TechDraw_View

semhustej, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request / idea discussion.**

When user desires to create elevations of a building, it is possible to select the building as a basis for TechDraw_View. This works well, but doors and windows have to be selected individually, because by default they are not included as children of Site/Building/Level container.

See sample dependency graph (windows are not children of wall or level or building):

[image]

**Proposed feature**
Enable user to create TechDraw_View of a Site/Building/Level by selecting only the main container.

Pros:

- would make it much easier to select the whole building for creation of techdraw views

Cons:

- maybe user is already overwhelmed by all of the options, how to enable BIM objects on a page. This will create another tool.

This idea was raised in this discussion, there are some more details there: https://forum.freecad.org/viewtopic.php?t=94392 
I am not completely sure if this is viable. Not because of technical aspect, but because the aspect mentioned in the Cons section.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-02-02:
We'd really like to know in ShapeExtractor that the Source object is from Arch/Draft and needs special handling.

The problem here is that there is no easy way (in c++ on the App side) to know that we have a Wall to deal with.  Module is "Part", TypeId is "Part::Feature".  

We can check for the presence of a Proxy property, but AFAICT we can't actually find out much about it.  That would tell us that the Source object belongs to a python module (Draft, BIM, etc)
but nothing beyond that.

kaiwas, 2025-02-03:
Perhaps the problem could be mitigated by a more convenient selection of elements in the 3D window.
https://github.com/FreeCAD/FreeCAD/issues/19381

Maybe some button in techdro (where we select a source) would be useful.

[image]

paullee0, 2025-02-03:
Cross-pointing from - https://forum.freecad.org/viewtopic.php?p=808943#p808943 :)

" Window objects are currently linked to Wall object by its Host property, technically making the former parents of the latter, rather than the apparent relationship implied in the tree view - it was an approach Yorik revert their relationship in earlier days to minimise cyclic dependency problem (and TNP?).

I have a look at TechDraw and realise it is probably largely c++, I can't help.

But a helper python would help by checking Wall.InList, then check which item(s) has/have host, and its host point back to the Wall - I think there had been a Draft function which already does this, e.g. to help in Ifc Export I guess. "

[image]

kaiwas, 2025-02-06:
I slightly modified the macro.
Algorithm for adding walls and windows to the section.
1) Click the source tab and clear the selection.
2) In the 3D window, hide unnecessary walls and activate the macro.
The macro selects all visible objects except spreadsheets/Varsets/groups/TechDraw elements (the list can be expanded in the macro).
3) Confirm the selection.

```
Macro code and video demonstration.

https://github.com/user-attachments/assets/a977c702-7584-4426-9d3e-30378e3d12b4

import FreeCAD as App
import FreeCADGui as Gui

# Get the active document
doc = App.ActiveDocument

# Function to check if an objects exlude
def is_exclude_objects(obj):
    return hasattr(obj, "TypeId") and obj.TypeId.startswith(("TechDraw::","App::VarSet","App::DocumentObjectGroup","Spreadsheet"))

# Collect all visible objects except TechDraw objects
visible_objects = []
for obj in doc.Objects:
    if obj.Visibility and not is_exclude_objects(obj):
        visible_objects.append(obj)

# Clear the current selection
Gui.Selection.clearSelection()

# Select all visible objects 
for obj in visible_objects:
    Gui.Selection.addSelection(obj)
```

I have not found a solution for how to select all windows and doors embedded in a wall. But I think the approach could be as follows:
1) Get a list of wall names.
2) Get the Host names on all objects in the document.
3) If the object's host names match the name of the wall list, add to the selection.
But my programming experience is very small.

yorikvanhavre, 2025-02-06:
There are two types of objects, in BIM, that don't appear in their parent's OutList: Windows (doors are actually a Window object too) and Rebars. Windows have a "Hosts" LinkList property that holds the walls, or any other object this window needs to make a hole in, and Rebars have a "Host" Link property that links to the structure (beam, column,...) that holds the rebar.

I would simply, at TechDraw level, look for a Hosts or Host property, and add what's in there...

kaiwas, 2025-02-06:
> There are two types of objects, in BIM, that don't appear in their parent's OutList:

yes. Thank you. During the search I noticed that the walls do not point to the windows/doors they contain.

Using a neural network and friends, we wrote a script to select all nested objects (and this is also convenient), but it does not work with walls.

In general, to summarize this discussion - a more convenient way to add elements to the source is needed. And there may be many solution options.

Just in case, a macro for selecting all objects inside the selected one with recursion. And also a script for inverting the selection of all objects.
I do not know what to do with this next, but for some tasks I will definitely use them. Probably something like this should be available by default without scripts.

https://github.com/user-attachments/assets/3bc0a099-bb6e-4784-a595-26840c9af222

Macros select "small tree"
```
import FreeCAD as App
import FreeCADGui as Gui

doc = App.ActiveDocument

def add_subobjects(obj):
    subobjects = obj.OutList

    for subobj in subobjects:
        if subobj.ViewObject.Visibility:
            FreeCAD.Gui.Selection.addSelection(subobj)

        add_subobjects(subobj)

selection = FreeCAD.Gui.Selection.getSelection()

for obj in selection:
    add_subobjects(obj)
```

Macros invert selecton
```
import FreeCAD as App
import FreeCADGui as Gui

# Get the active document
doc = App.ActiveDocument

if doc:
    # Get all objects in the document
    all_objects = doc.Objects
    # Get the currently selected objects
    selected_objects = Gui.Selection.getSelection()

    # Invert the selection
    for obj in all_objects:
        if obj in selected_objects:
            Gui.Selection.removeSelection(obj)
        else:
            Gui.Selection.addSelection(obj)

    App.Console.PrintMessage("Selection inverted.\n")
else:
    App.Console.PrintError("No active document found.\n")
```

Syres916, 2025-02-06:
> Using a neural network and friends, we wrote a script to select all nested objects (and this is also convenient), but it does not work with walls.

That's because you are not using QTreeWidget, it's not much more effort, for me the most painful part is the numerous ways the tree can be a child of MainWindow so something is required like:

```
    modelview = mw.findChild(QtGui.QDockWidget,'Model')
    if modelview: # standard Combo View after 0.21
        tree = modelview.findChildren(QtGui.QTreeWidget)[0]
    else:
        treeview = mw.findChild(QtGui.QDockWidget, 'Tree view')
        if treeview: # Treeview and Properties view after 0.21
            tree = treeview.findChildren(QtGui.QTreeWidget)[0]
        else: # 0.21 and before Combo view
            tab = mw.findChild(QtGui.QTabWidget, "combiTab")
            tree = tab.widget(0).findChildren(QtGui.QTreeWidget)[0]
    top = tree.invisibleRootItem().child(0)
```

Hopefully I'll have some time on Friday morning to get a reasonably polished macro and source code to go into Techdraw.

Syres916, 2025-02-06:
@paullee0 the concern I have with only using attributes is arrays of windows, surely in an office block scenario, the user wouldn't create individual windows would they?

kaiwas, 2025-02-06:
> 'apparent children' objects of a Wall is indicated in my post and Yorik above.

Sorry, I didn't notice (translation problem).
I'll try to experiment with this code :)

paullee0, 2025-02-06:
> [@paullee0](https://github.com/paullee0) the concern I have with only using attributes is arrays of windows, surely in an office block scenario, the user wouldn't create individual windows would they?

Yes, arrays of Window, or CurtainWall.  Not sure I understand what is the problem?

kaiwas, 2025-02-07:
> Select a Wall object, run below code -

yeah. programming is clearly not my strong point.
Yesterday I spent the whole evening trying to figure out how to add windows from several selected walls to the selection (now only the first selected one is analyzed). Nothing worked, although there probably needs to be some simple cycle added there

yorikvanhavre, 2025-02-07:
@Syres916 About arrays: That's another can of worms :) There is a multitude of ways one can, for ex, make a series of similar windows in a building. The BIM WB itself is a bit fuzzy in that regard, as it favors clones, but more and more people are using Draft arrays, that use links.

Probably the way forward here would be to discard the old cloning system and make everything use links instead. But I would say, let's go step by step, and solve first the door/window problem here, then we see about clones/links. Some of this is probably already supported by TechDraw anyway.

kaiwas, 2025-02-07:
> About arrays:

By the way, since you mentioned arrays. Maybe there is a way to add a simplified array view that only works on one axis.

Or at least set the default quantity values ​​to 1.
I constantly have to change the values ​​every time I use the array tool. (The quantity value of 2 for the Y-axis is very inconvenient when I have to build an array in the horizontal plane)

https://github.com/FreeCAD/FreeCAD/issues/16865

yorikvanhavre, 2025-02-07:
This really should be in another discussion, but I imagine that more as an UI problem: How to make the array tool show a "simplified" setup for 1-axis arrays, etc.

Syres916, 2025-02-08:
I have a very alpha macro, assumptions:

1) The user doesn't use the same label for multiple objects (no lazy users here :smile:)
2) Only one object can be selected at present, it's been fighting me but I think I can sort it in the next 24 hours.

Code:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
__title__ = "BIMSelectAllVisible.py"
__author__ = "TBA"
__version__ = "1.0"
__date__ = "2025/02/09"  # YYY/MM/DD

__Comment__ = ""
__Help__ = "Select objects"
__Status__ = "alpha"
__Requires__ = "freecad 0.21 (not tested in previous versions)"

# import required libraries
import FreeCAD as App
import FreeCADGui as Gui

# from BIM.importers import exportIFC

import PySide
from PySide import QtGui, QtCore
from PySide.QtGui import *
from PySide.QtCore import *

doc = App.ActiveDocument.Name
mw = Gui.getMainWindow()

selectedList = []
addToSelectionList = []

print("============================================")

# try to get internal object associated with QTreeWidgetItem
def getInternalObject(treeItem):
    # first looks by Label
    obj = App.ActiveDocument.getObjectsByLabel(treeItem.data(0, 0))
    if len(obj) > 0:
        return obj[0]
    else:
        # if not found, looks by Name
        return App.ActiveDocument.getObject(treeItem.data(0, 0))
    pass

# for each object in treeview
def traverse(list, pad):
    # for each child
    for idx in range(list.childCount()):
        item = list.child(idx)
        name = item.data(0, 0)
        obj = getInternalObject(item)
        if (
            not obj.Name in addToSelectionList
            and obj.TypeId != "App::DocumentObjectGroup"
            and hasattr(obj, "Visibility")
            and obj.Visibility == True
        ):
            addToSelectionList.append(obj.Name)
        traverse(item, pad + "  ")

# for each object in treeview
def traverseFindSelected(list, pad):
    # for each child
    for idx in range(list.childCount()):
        item = list.child(idx)
        name = item.data(0, 0)
        obj = getInternalObject(item)
        if not obj.Label in selectedList:
            traverseFindSelected(item, pad + "  ")
        break
    traverse(item, pad + "  ")

experimentalCheck = True

if experimentalCheck:

    prefs = App.ParamGet("User parameter:BaseApp/Preferences/Document")
    if prefs.GetBool("DuplicateLabels", False):
        App.Console.PrintWarning(
            "This macro does not work with documents having multiple objects with the same label\n\n"
        )

    modelview = mw.findChild(QtGui.QDockWidget, "Model")
    if modelview:  # standard Combo View after 0.21
        tree = modelview.findChildren(QtGui.QTreeWidget)[0]
    else:
        treeview = mw.findChild(QtGui.QDockWidget, "Tree view")
        if treeview:  # Treeview and Properties view after 0.21
            tree = treeview.findChildren(QtGui.QTreeWidget)[0]
        else:  # 0.21 and before Combo view
            tab = mw.findChild(QtGui.QTabWidget, "combiTab")
            tree = tab.widget(0).findChildren(QtGui.QTreeWidget)[0]
    top = tree.invisibleRootItem().child(0)

    # get selected objects
    sels = Gui.Selection.getSelection()
    if not sels:  # not sels
        App.Console.PrintWarning("Select something first!\n\n")
    if len(sels) > 1:
        App.Console.PrintWarning("Only select one object!\n\n")
    else:  # sels
        for obj in sels:
            currentSelectedName = obj.Label
        App.Console.PrintLog(
            "Selecting all visible objects of "
            + doc
            + " - "
            + currentSelectedName
            + "\n"
        )
        # find the treeitem for ActiveDocument
        for idx in range(top.childCount()):
            doc = top.child(idx)
            if doc.data(0, 0) == App.ActiveDocument.Label:
                top = doc

        # store a list of selected items
        for sel in tree.selectedItems():
            current = sel
            selectedList.append(current.data(0, 0))

        traverseFindSelected(top, "")

for newSel in addToSelectionList:
    Gui.Selection.addSelection(App.ActiveDocument.Name, newSel)

print("Finished")

```

Result (you'll notice the filename):

[image]

Syres916, 2025-02-09:
Added into source locally:

[image]

semhustej, 2025-02-17:
I was surprised, that when selection base objects for Arch_SectionPlane, I could select only the wall object and in the resulting Shape2DView or BIM_view on a TechDraw page, the windows and doors, which are part of the selected wall, are taken into account. It might be helpful to look into how Arch_SectionPlane is selecting depending doors and windows so that they are displayed.

Shape2DView:

[image]

BIM_View on a TechDraw page:

[image]

yorikvanhavre, 2025-02-18:
The Draft/BIM tools have Python methods to select all dependent BIM objects (`draftutils.groups.get_group_contents()`).
It's probably not a good idea to reproduce that in the TechDraw C++ code as it would duplicate stuff. It's also annoying to setup a temp interpreter to use the Python function.

To address the problem stated in OP, I think I would just do this where the TechDraw View gathers the objects to be shown:

- Scan for objs with Host or Hosts properties in Inlist
- Add those whose Host/Hosts contains the object

Probably in https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/TechDraw/App/DrawViewPart.cpp#L193 but I can ´t find a place where that tool gathers child objects... I might be missing something..