FreeCAD/FreeCAD #19371: [Gui] [3D Viewport] Mutliple issues with visibility of the Frames Per Second / Response text display

Syres916, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm sure this has already been logged but couldn't find it. Here are examples of the visibility issues with the FPS/responsiveness text output in the 3D viewport:

Color doesn't change according to the background affecting light colored backgrounds
[image]

Depending on the theme, when the overlay is enabled with the left dock in use the Property tabs obscure part of the text
[image]

[image]

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Light-modern.qss/Light-modern/Fusion
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * CfdOF 1.29.6
```

### Subproject(s) affected?

Core

### Anything else?

I'm working on a proposal and will publish shortly when I get all combinations of overlay and background colors determined.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-02:
The branch for the correction is [here](https://github.com/Syres916/FreeCAD/tree/FPS_Fixes_dev) , example of change using the Axis Letter Color form the coordinate system in the bottom right hand corner (by default):

[image]

[image]

[image]

[image]

It might look as though the text has been moved too far to the right but there are some bizarre scenarios with the way the overlay can be docked usually by accident which came up in testing.

If the Axis Letter Color is not acceptable the ColorBarText color could be used as an alternative but as the FPS and coordinate system are in the same section of Preferences this seemed logical and I can enhance the Tooltip to specify it applies to both.

Syres916, 2025-02-02:
FYI  @FreeCAD/design-working-group

Kastroz, 2025-02-02:
Also using moving in 3d space the 3dconexion Spaceball image serious trouble Nvidia GeForce GTX 1050 Ti

Syres916, 2025-02-02:
@Kastroz please raise that as a separate issue, I don't have the test kit to be able to fix it.

Syres916, 2025-02-03:
@maxwxyz I really think this was a bug introduced by the Overlay as there was never an issue seeing all the text in 0.21.x.

PR #19385 (2025-02-11): [Gui] fix the FPS text visibilty issues raised in #19371
Result of this fix can be seen in https://github.com/FreeCAD/FreeCAD/issues/19371#issuecomment-2629540185

Fixes #19371
closed 2025-02-11 by chennes