FreeCAD/FreeCAD #19372: FreeCAD Light/Dark: Undo/Redo position chevron makes it difficult to run the command

furgo16, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Notice how the Undo/Redo buttons in FreeCAD Light/Dark are narrower than all other buttons that have a chevron to expand a dropdown to select multiple items in the list. Compare also with FreeCAD Classic:

FreeCAD Dark:
[image]

FreeCAD Classic:
[image]

Since there is no space for the chevrons on the right hand side of the buttons, they are put on the lower right corner of the button.

With the narrow button, as the chevron occupies a big portion of the button itself, it's really difficult to hit undo/redo and not expand the dropdown. As a result, an operation that generally takes one click ends up taking two or more. E.g.: first attempt to click: dropdown expanded by mistake, second click to close the dropdown, third click to run the command that was intended with the first click.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Snap 1291
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * parts_library
  * Arch
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
@MisterMakerNL FYI, workaround https://github.com/FreeCAD/FreeCAD/issues/19129 seems not ideal

furgo16, 2025-02-04:
@MisterMakerNL great, thanks! One thing I've noticed on that screenshot is that the chevron icon is a bit blurry (or pixelated). Is that intended?

furgo16, 2025-04-10:
@MisterMakerNL has the fix been added to https://github.com/FreeCAD/FreeCAD/pull/19367? If it has, would it be possible to extract the fix and submit a separate hotfix PR? If not, I'd be willing to submit a PR for it.

Happy to do the work, I just need some pointers. Thanks!

MisterMakerNL, 2025-04-10:
Oof it was fixed but it is again a bit broken after the switch to svg files...

[image]

But anyway for the non-svg files this should work:

```
/* undo button */

QToolBar[objectName="Edit"] > QToolButton {
  padding-right:14px;
}

QToolBar[objectName="Edit"] > QToolButton::menu-arrow {
  background-image: url(qss:images_classic/arrow-down-lightgray.png);
  background-position: center center;
  background-repeat: none;
  subcontrol-origin: padding;
  subcontrol-position: center center;
  height: 6px;
}

QToolBar[objectName="Edit"] > QToolButton::menu-arrow:hover {
  background-image: url(qss:images_classic/arrow-down-white.png);
}

QToolBar[objectName="Edit"] > QToolButton::menu-arrow:open {
  background-image: url(qss:images_classic/arrow-down-white.png);
}
```

furgo16, 2025-04-11:
@maxwxyz can we unlink https://github.com/FreeCAD/FreeCAD/pull/19367 from this issue? Thanks!

closed 2025-04-15 by kadet1090