FreeCAD/FreeCAD #19374: Draft: cannot select objects under a layer if the layers are inside a Part container

furgo16, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For architectural drawings, and during the 0.22 release cycle, I used to import DXF floor plans with layers into FreeCAD and then use [Std_Part](https://wiki.freecad.org/Std_Part) containers to put them in. With the Part container, I could then easily set the placement for the floor plans in one place.

I've recently opened a document I created during 0.22, and tried to add a floor plan to it following the same approach. Importing the top layers object works well, but as soon as I put that layers object inside a Part parent, the objects inside the layer become unselectable.

Steps to reproduce: 

1. Download and uncompress [Part_layers_bug.zip](https://github.com/user-attachments/files/18634443/Part_layers_bug.zip)
2. Open that test file with FreeCAD
3. Navigate to the `P0_10_edi_sani` layer and expand it
4. Try to select one of the objects in that layer by clicking on them. It doesn't work => NOK
5. In the document tree, move the `Layers` object to the root of the document (move it out of its `Part` container)
6. Navigate to the `P0_10_edi_sani` layer and expand it
4. Try to select one of the objects in that layer by clicking on them. It works => OK

The test file is a stripped out version of a floor plan I've been working with. It was originally imported with the C++ importer (if I remember it correctly), with all objects created as `Part:PartFeature`. If that makes a difference, I'll mention they were then converted to Draft objects in the Python console by sending each layer to the console and running `Draft.upgrade(obj.Group, delete=True, force="draftify")`.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Snap 1291
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * parts_library
  * Arch
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-02-03:
I can reproduce the issue following steps from OP. When trying to select object inside of layer, I get this error message:
`1e-07 Selection.cpp(1518): Sub-object Part_layers_bug#Part.LayerContainer.Layer007.Wire051. not found`

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

furgo16, 2025-02-03:
Good catch, thanks. I get the error too. It comes from the C++ side, here:

https://github.com/FreeCAD/FreeCAD/blob/34d9ac00506ace1b930a4efa075b97bee810f1c7/src/Gui/Selection.cpp#L1518

But I can't figure out why.

furgo16, 2025-02-03:
I've noticed this on my original document: the Part components that contained layers had all elements from the layer listed as part of the Part's `Group` property. On the test file on this issue, only the layers are listed, but not the layer elements. I don't know exactly why. Even if I remove all layers but keep the objects, then recreate the layers and move the objects to them, once adding the `Layers` object to `Part`, the objects will not be added to `Part`'s `Group` property.

In any case, seeing that, I've found out a workaround:

1. Select the `Part` object on the document tree
2. Select its `Group` property in the Property editor
3. Click on the ellipse button
4. On the dialog to pick group members, add all the `WireNNN`, `ArcNNN` and `LineNNN` objects
5. Press <kbd>OK</kbd>
6. Now the items are selectable as expected

maxwxyz, 2025-02-03:
@Roy-043 FYI

Roy-043, 2025-02-05:
This actually a Core issue. Probably related to the `"App::PropertyLinkListHidden"` property type used for the `Group` property of the layer.

Roy-043, 2025-07-10:
What we see here has two causes:
1. A Std_Part is a `GeoFeatureGroup` which normally should include the objects in nested non-geo groups.
2. The Group property of a Draft_Layer is of the type `PropertyLinkListHidden` (since V1.0).

When you drag a normal group with content into a Std_Part, the group and its (recursive) content are added to the part's Group. When you drag a LayerContainer into a Std_Part this mechanism does not work for the content of the layers because of their Group's 'hidden' property type.

As already mentioned, you can manually add the content of the layers to the part's Group to solve this.

Alternatively you can create the same issue for a normal group in a Std_Part by manually removing all objects, except the nested group itself, from the part's Group.

BTW: This interaction between geo groups and non-geo groups is the reason why you should use the `addObject` method to add objects to groups in Python.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/8acec2cb5c006f39cdce8c82b73c59427152d23c

The tree checks whether the Part can resolve the layer child before building that selection path, avoiding the sub-object not found error.