FreeCAD/FreeCAD #19375: FEM: Errors when Thermal Expansion Reference Temperature is taken right from the materials database

FEA-eng, 2025-02-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The following error messages appear when the newly introduced FEM material property named Thermal Expansion Reference Temperature is taken straight from the materials database (i.e. material is only selected from the database without checking the "Use this task panel" box).

For example:
1. Create Part Cube
2. Add an Analysis container.
3. Mesh the cube.
4. Add solid material and choose e.g. CalculiX-Steel without changing anything else in the task panel.
5. Add initial temperature.
6. Change the analysis type to thermomech.
7. Try running the analysis.

```
00:21:06  Unexpected error when writing CalculiX input file: 'ThermalExpansionReferenceTemperature'
00:21:06  Running the Python command 'FEM_SolverRun' failed:
Traceback (most recent call last):
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femcommands\commands.py", line 1136, in Activated
    run_fem_solver(self.selobj)
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femsolver\run.py", line 110, in run_fem_solver
    fea.run()  # standard, no working dir is given in solver
    ^^^^^^^^^
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 612, in run
    self.write_inp_file()
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femtools\ccxtools.py", line 379, in write_inp_file
    self.inp_file_name = inp_writer.write_solver_input()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femsolver\calculix\writer.py", line 167, in write_solver_input
    write_femelement_material.write_femelement_material(inpfile, self)
  File "D:\FreeCAD_weekly-builds-40041-conda-Windows-x86_64-py311\Mod\Fem\femsolver\calculix\write_femelement_material.py", line 85, in write_femelement_material
    mat_obj.Material["ThermalExpansionReferenceTemperature"]
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

'ThermalExpansionReferenceTemperature'
```

Just checking the "Use this task panel" box in step 4 prevents the issue.

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/113.268
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

marioalexis84, 2025-02-03:
No material with thermal model has ThermalExpansionReferenceTemperature property defined

FEA-eng, 2025-02-03:
> No material with thermal model has ThermalExpansionReferenceTemperature property defined

Yes, but shouldn't it be just set to 0 by default in such cases? Or at least the error message could be more user-facing. I guess it may apply to other material properties as well.

marioalexis84, 2025-02-03:
Indeed, it applies to all properties.
I think it's better to return a `nan` value (`Base::Quantity` supports this).
We can temporarily add a try/except block to check the property until we add a more general solution.

PR #19424 (2025-02-05): Fem: Check reference temperature - fixes #19375
Fixes #19375
@maxwxyz Without this check, thermal analysis is unusable.

@FEA-eng
commit 703a1b52 (2025-02-05): Fem: Check reference temperature - fixes #19375
closed 2025-02-05 by maxwxyz