FreeCAD/FreeCAD #19379: Segmentation Fault When Removing Dragger Callback (Local vs. Global Variable Issue)

HakanSeven12, 2025-02-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using an SoTranslate2Dragger in FreeCAD, removing the dragger callback via removeDraggerCallback results in a segmentation fault (SIGSEGV) under certain conditions. Specifically, if the dragger is defined as a local variable inside a function and then referenced in another function for callback removal, the crash occurs. However, when the dragger is declared as a global variable, the removal works as expected.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=93627

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * FreeCAD-Ribbon 1.5.1.2
  * Road 2025.1.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-03:
@kadet1090 you have some experience with draggers, any idea?

luzpaz, 2025-09-26:
@HakanSeven12 is this still the case ?

HakanSeven12, 2025-09-26:
I will test it again

HakanSeven12, 2025-09-27:
Same problem still exist