FreeCAD/FreeCAD #19381: Add the ability to "select all visible" (in the 3D window).

kaiwas, 2025-02-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

For example.
When we create a view in Tech Draw and add new elements, we have to add them manually. We can add each new element with a single selection, or we can select everything with a frame in the 3D window, but for this we constantly have to move the panel aside.
If we had the ability to select everything visible through a button in the menu, this problem would be solved faster.

Video demonstration

https://github.com/user-attachments/assets/5d6b9d54-5d59-4e71-8b9a-28d97e11f144

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/91.7938
Installed mods: 
  * FreeCAD_SketchArch
  * lattice2.backup1738303181.7780895 (Disabled)
  * lattice2
  * PitchedRoof
  * Road 2025.1.16
  * addFC 1.2.4
  * freecad.gears 1.3.0
  * sheetmetal 0.7.5
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-02-03:
You can find macro **SelectVisible** in Addon Manager
https://github.com/FreeCAD/FreeCAD-macros/blob/master/Utility/SelectVisible.FCMacro

```
import FreeCAD as App
import FreeCADGui as Gui

doc = App.activeDocument()

if not doc:
    App.Console.PrintWarning('SelectVisible: no active document')
else:
    Gui.Selection.clearSelection()
    for o in doc.Objects:
        if o.ViewObject.Visibility:
            Gui.Selection.addSelection(o) 
```

kaiwas, 2025-02-03:
> You can find macro **SelectVisible** in Addon Manager

Yes. I know that the problem can be solved with macros. But such a possibility should be available by default (especially since it is very easy to implement)

kaiwas, 2025-02-03:
I found the command "Select visible objects" in the "View" -> "Visibility" menu.
But this command also selects all visible objects (including those located outside the 3D window)

https://github.com/user-attachments/assets/b61e86ea-4baa-454e-914a-f2020c289e42

tarman3, 2025-02-03:
PS Which application do you use for show keys while record screen?

kaiwas, 2025-02-03:
@maxwxyz I changed the title and corrected the description because I found that the program has the necessary button (though it works a little differently than expected).

tarman3, 2025-02-03:
> screenkey (linux)

X11 ??

kaiwas, 2025-02-03:
> X11 ??

Yes