FreeCAD/FreeCAD #19389: 3D navigation using 3DConnexion SpaceMouse is now shaky on Win11

karamellpelle, 2025-02-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I tried the newest developer version, and now the 3D navigation using SpaceMouse is literally shaky on Windows 11. I read on the blog that there was a newly a commit for MacOS (https://github.com/FreeCAD/FreeCAD/commit/836ad91f3ddd3ec63b84b8e3f87a767d5465261d), and I expect this to have caused the problem on Windows some way. What do you think, @pskowronskiTDx ?

----

PS: I have not gotten Parkinson's disease during the last weeks

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * IconThemes
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pskowronskiTDx, 2025-02-03:
This behaviour is expected due to inproper implementation of the runtime switch: https://github.com/FreeCAD/FreeCAD/issues/19223. It looks like "Both" option has been already enabled for weekly builds. @chennes Can you confirm?

chennes, 2025-02-03:
This will be fixed by https://github.com/FreeCAD/FreeCAD/pull/19226

karamellpelle, 2025-02-03:
Sorry, but I used the search functionality but didn't find the referred issue :( I guess it was because the issue was closed.

closed 2025-02-03 by karamellpelle