FreeCAD/FreeCAD #19392: [Problem] CAM: Adaptive Profiling does not honor stepover value

rsiadmin2015, 2025-02-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Although Adaptive Clearing is maturing, it appears that the Adaptive "Profiling" mode does not honor any of the parameters in the Operations pane. Specifically, it appears that stepover is hardcoded to some portion of tool diameter. Based upon the posted output, the stepover appears to be appx 72% - 18.045mm of the tool diameter....

Loading a 25mm diameter tool, setting Operation Type parameter to "Clearing" and changing Step Over Percent parameter, does change the stepover in a computed path.
Changing the Operation Type parameter to "Profiling" computes a path that appears to be a fixed portion of the tool diameter. Changing the Step Over Percent parameter and recomputing has no effect on output.

Based upon the action when Step Over Percent is used with the "Clearing" type, it was expected that a similar calculation would occur with the "Profiling" type wherein the smaller the stepover, the more profile paths at that calculated percent of tool diameter would be created.

Screenshot attached to indicate output.

Regards,
Ted.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * btl 0.9.9
  * CamScripts
  * Curves 0.6.54
  * Defeaturing 1.2.3
  * fasteners 0.5.37
  * FeedsAndSpeeds 0.6.0
  * FrameForge 0.1.2
  * freecad.gears 1.3.0
  * freecad.optics_design_workbench 0.1.5
  * GDT
  * ImportNURBS 1.1.0
  * InventorLoader 1.5.1
  * MeshRemodel 1.10.36
  * pyrate (Disabled)
  * Render 2024.12.15
  * sheetmetal 0.7.5
  * toSketch 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-02-06:
Could you please add an example file and test if it works in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

rsiadmin2015, 2025-02-10:
[adaptive_profile_test_001.zip](https://github.com/user-attachments/files/18737693/adaptive_profile_test_001.zip)

Max - also tried it out in dev version as requested; the results are very inconsistent value-to-value (no specific pattern detected in stepovers) - however in forcing the accuracy to full (due to finding a value combination that produced lots of pythinic errors - mostly by nulling out helix values), I was able to find a range of useful stepovers (in the 20-35% range, which is where I want my chipload to be) but back in the CLEARING mode which then went from a HSM/Trochoidal-like path to a profiling-like path. 
I'm fine with that, and I realize that Adaptive is noted as "quirky" still, so I'm not going to push on it. I wasn't able to find a really good "here's what to expect from FC Adaptive" compared to other CAM systems (I'm quite familiar with SprutCam adaptive/HSM, MasterCam 3d Adaptive, and AutoDesk's offerings) but to make it easy, I'll simply change my vocabulary to fit FC instead.

closed 2025-02-10 by rsiadmin2015