FreeCAD/FreeCAD #19396: Sketcher TrimEdge deletes far too many constraints in 40041, Works as expected in 1.0.

casterle, 2025-02-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Using FC 1.1.0dev.40041, I am trying to trim the edges between the two horizontal lines in the attached project. 

The sketch is fully constrained before the trim is applied:

[image]

After trimming, the sketch has 62 degrees for freedom; ALL of the constraints have been removed:

[image]

I saved the project and opened it in the 1.0 release version, and the trim worked as expected.

[Exercise 7 - Copy.zip](https://github.com/user-attachments/files/18649445/Exercise.7.-.Copy.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: Dracula.qss/FreeCAD Dark/Qt default
Logical/physical DPI: 96/93.3153
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * boltsfc 2022.11.5
  * CurvedShapes 1.0.13
  * Curves 0.6.55
  * Defeaturing 1.2.3
  * Dracula 0.0.5
  * DynamicData 2.71.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * Freecad-Built-in-themes-beta 1.2.2
  * freecad.gears 1.3.0
  * Gridfinity 0.8.2
  * Help 1.0.3
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * MeshRemodel 1.10.36
  * OpenDark 2024.3.13
  * parts_library
  * ProDarkThemePreferencePack 1.0.0
  * SelectorToolbar
  * sheetmetal 0.7.5
  * Silk 0.1.5
  * STEMFIE 0.3.1
  * ThreadProfile 1.92.0
  * toSketch 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-02-04:
the regression was introduced sometime in early December 2024, last good build I have is:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.5
  * CfdOF 1.29.6
```

First bad build I have is:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.5
  * CfdOF 1.29.6
```

Syres916, 2025-02-04:
@AjinkyaDahale Git Bisect has determined that https://github.com/FreeCAD/FreeCAD/commit/263331f6fb6c80416bc0d5f3560b6a39c9e50485 causes this regression, could you please have a look. The extra sea of red in the Report View isn't really what a user expects to see:

```
Sketcher constraint number 9 is malformed!
The Sketch has malformed constraints!
Equality constraints between line and arc are not supported.
Sketcher constraint number 34 is malformed!
The Sketch has malformed constraints!
Equality constraints between line and arc are not supported.
Sketcher constraint number 34 is malformed!
Sketcher constraint number 35 is malformed!
The Sketch has malformed constraints!
```

AjinkyaDahale, 2025-02-07:
Please check if #18916 fixes the issue.

maxwxyz, 2025-02-10:
Please test with the new weekly tomorrow: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

casterle, 2025-02-11:
1.1.0dev.40176 changes the behavior. If I trim the right arc first, I get 19 constraints listed as over-constrained. Trimming the other arc  does not change the over-constrained count. 

If I instead trim the left arc first, everything is fine until I trim the right arc at which point the same 19 are constraints are listed as over-constrained.

This is using the model I originally uploaded.

AjinkyaDahale, 2025-02-16:
It's the symmetric constraint that's causing the major issue. This results in an implicit point on object constraint, and the trim makes an additional external one. Then there's an equality constraint that is also probably affected by this. If we remove the symmetric constraint and one of the equality constraints we get a fully constrained object.

casterle, 2025-02-17:
> It's the symmetric constraint that's causing the major issue. This results in an implicit point on object constraint, and the trim makes an additional external one. Then there's an equality [...]

I can't reproduce this (using 1.1.0dev.40239). Did you remove the constraints before you did the trim? Which equality constraint did you remove? Did you remove the symmetry constraint before or after the equality constraint?

Note that this does not happen in 1.0 - I don't get the over-constraints, and end up with one DOF. Perhaps the sketcher refactoring changed this behavior?

IMHO, regardless of the cause, I think this is something the sketcher should handle on its own - the user shouldn't have to understand how the sketcher works internally to avoid these kinds of issues.

AjinkyaDahale, 2025-02-17:
> I can't reproduce this (using 1.1.0dev.40239). Did you remove the constraints before you did the trim? Which equality constraint did you remove? Did you remove the symmetry constraint before or after the equality constraint?

Removed constraints after trim. Don't recall which equality constraint I deleted, likely the first. Symmetric before equality, IIRC.
 
> Note that this does not happen in 1.0 - I don't get the over-constraints, and end up with one DOF. Perhaps the sketcher refactoring changed this behavior?

It indeed did. We need to adjust symmetric constraints as well.

> IMHO, regardless of the cause, I think this is something the sketcher should handle on its own - the user shouldn't have to understand how the sketcher works internally to avoid these kinds of issues.

Agreed. At least if the sketch is fully constrained.

AjinkyaDahale, 2025-02-18:
I tried to create a simpler example where this happens, and it seems the problem may be in the treatment of constraints in the first place. In the following example, try to create a symmetric constraint between both ends of the same curve and the x-axis.

[minimal-with-symmetric.zip](https://github.com/user-attachments/files/18852833/minimal-with-symmetric.zip)

@casterle could you share the file just after the trim in 1.0? I would like to see the difference between the new constraints in both versions.

casterle, 2025-02-18:
> [@casterle](https://github.com/casterle) could you share the file just after the trim in 1.0? I would like to see the difference between the new constraints in both versions.

Gladly!

[Exercise 7 - Post Trim in 1.0.zip](https://github.com/user-attachments/files/18854855/Exercise.7.-.Post.Trim.in.1.0.zip)

casterle, 2025-02-18:
@AjinkyaDahale I just realized that you're the guy doing the sketcher refactoring. Thank you! You and the other devs have made FreeCAD much, much better.

AjinkyaDahale, 2025-03-04:
@casterle thank you for your kind words, and sorry for the belated response and yet belated fix.

AjinkyaDahale, 2025-03-06:
@casterle would you be able to provide the resulting file after the trimmings, done in 1.0, for the file in #19425 as well? It has a similar issue.

casterle, 2025-03-07:
Sorry for the late reply. Do you want me to try to trim the 3/4 that the OP was trying to delete?

AjinkyaDahale, 2025-03-07:
@casterle yes please. Please attach the file in that issue.

casterle, 2025-03-07:
> [@casterle](https://github.com/casterle) yes please. Please attach the file in that issue.

Glad to help.

[trim_edge_issue.FCStd.zip](https://github.com/user-attachments/files/19134565/trim_edge_issue.FCStd.zip)
[trim_edge_issue_Post_Trim.FCStd.zip](https://github.com/user-attachments/files/19134564/trim_edge_issue_Post_Trim.FCStd.zip)

imaluschi, 2025-03-26:
As I understand this topic, it is still not fixed.

[image] [image]

[image] [image]

[image]

AjinkyaDahale, 2025-03-30:
@imaluschi please check with #20150.

closed 2025-04-10 by hyarion

gianlucafiore, 2025-04-27:
It happens to me that my freecad crashes and closes: 

[image]

[file.zip](https://github.com/user-attachments/files/19929658/file.zip)

luzpaz, 2025-04-27:
> 41354

@gianlucafiore can you test on a more current 1.1dev ? 

```
$ git rev-list --count HEAD                                                                                                                     
41446
```

leoheck, 2025-04-29:
With the latest weekly I was able to trim the `Exercise 7` without having the constraints removed.
It looks good. Do you also condirm @casterle ?

See:

[image]

[image]

```c++
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.3
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.0

```

leoheck, 2025-04-29:
ah, sorry, this was closed already, damm GitHub filter. hahah